esphome/feature-requests #2201: FastLED and/or NeoPixelBus on a EPS32 board with esp-idf framework

roykrikke, 2023-04-10:
**Describe the problem you have/What new integration you would like**
I'd like it if ESPHome could support FastLED and/or NeoPixelBus on a EPS32 board with the esp-idf framework 
Board example e.g. like [Atom Lite](https://docs.m5stack.com/en/core/atom_lite)

**Please describe your use case for this integration and alternatives you've tried:**
There are several [ESP32 based boards](https://docs.m5stack.com/en/products) designed which could be used with FastLED and/or NeoPixelBus for controlling e.g. SK6812 or WS2812 led strips, it would be great if they could be run with ESPHome

**Additional context**
```
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: recommended

light:
  - platform: neopixelbus
    id: ${id_name}_light
    name: ${friendly_name} Light
    pin: GPIO27
    chipset: SK6812
    num_leds: 1
    rgb_order: GRB
```

randybb, 2023-10-31:
[ESP32 RMT](https://esphome.io/components/light/esp32_rmt_led_strip) is a native peripheral, so just use it instead of fastled/neopixelbus which are arduino libs.

ssieb, 2023-10-31:
Why do you need to use esp-idf?
The libraries require Arduino, so there's nothing esphome can do about it.  As mentioned just now, use the RMT component instead.

closed 2023-10-31 by ssieb