esphome/feature-requests #2202: Support for HC-12 433mhz Transmitter / Receiver

deanfourie1, 2023-04-11:
**Describe the problem you have/What new integration you would like**
Integration for the HC-12 433mhz Transceiver module.

**Please describe your use case for this integration and alternatives you've tried:**
other units will not cover the range or distance I need to transmit at.

**Additional context**
Great module, communicates over uart. Would be great to see this on the ESPhome list!

Zaleo80, 2024-03-21:
Looking to do this myself, not jet started on the software part, but is this not coverd with the uart component?

https://esphome.io/components/uart.html

madurapa, 2025-02-19:
I have made an external component for communicating with HC-12. leave it here if anyone is interested.

https://github.com/madurapa/esphome-hc12