esphome/feature-requests #2203: Support for 128X64 I2C ST7567S Displays

Aikhjarto, 2023-04-11:
**Describe the problem you have/What new integration you would like**

I would love to see support for 128X64 I2C ST7567S displays in ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**

With https://github.com/luetee/ST7567S_128X64_I2C there is already a demo-project, for this display for Arduino, which did work with my display.

**Additional context**

Link to controller's datasheet: https://www.buydisplay.com/download/ic/ST7567S.pdf

atto84, 2023-04-13:
also please Thank you

markuspulch, 2023-07-13:
+1 those displays are really interesting. So support within esphome would be awesome!
Is there an option to sponsor such feature requests?

markuspulch, 2023-11-07:
@Aikhjarto @atto84  just found this one but didn't tested so far... https://github.com/nicolaij/esphomes

atto84, 2023-11-07:
Thank you. I will test. Odesláno z iPhonu7. 11.

latonita, 2023-12-14:
@Aikhjarto @atto84 
I have just made a component here https://github.com/latonita/esphome-st7567s-display

please try

latonita, 2023-12-20:
@Aikhjarto @atto84 @markuspulch 
Component i linked in previous comment is for current version (2023.11), 
are you happy to test component i've made for esphome official repo for next ver?

It's as simple as add it to your config (you need to install dev version of esphome)
```
  # equivalent shorthand for GitHub pull request
  - source: github://pr#2639
    components: [ st7567_base, st7567_i2c, st7567_spi ]
```

closed 2023-12-27 by clydebarrow