esphome/feature-requests #2206: Add partial update support with deep sleep for Inkplate

AlexanderGlogger, 2023-04-16:
**Describe the problem you have/What new integration you would like**

When Inkplate was in deep sleep the display performs a full refresh every time it wakes up. I don’t want that, I want it to full refresh only every so many updates (Defined by "full_update_every"), with deep sleep in between. Even if the inkplate wakes up from deep sleep mode.

**Please describe your use case for this integration and alternatives you've tried:**

When running an inkplate display on battery, as much energy as possible must be saved. For this purpose, the inkplate should always go into deep sleep mode after the data has been updated and wake up again after some time, and display new values again. However, this means that a full refresh is always performed. For use cases where the screen is very rarely updated, that might be the best option, but for my application it would be nice to be able to perform partial updates.

**Additional context**

[https://github.com/SolderedElectronics/Inkplate-Arduino-library/commit/c25f07915bac62e7c63789d347b3015cc97b5c5e](url)

The Inkplate Arduino Library already has this functionality implemented. However, since I don't have the skills, I can't add this functionality to the esphome inkplate componente myself. However, I am of course available for testing.

werdnum, 2025-01-03:
Came across this. I'm by no means an expert but just leaving some drive by notes to save somebody some time:

The commit @AlexanderGlogger referenced shows the rough approach as follows:

1. Put the screen back to the way it was before but don't call `.display()`.
2. Call `.preloadScreen()` to copy from the 'new' buffer to the 'partial' buffer.
3. Call `.clearDisplay()` and write the new data.
4. Call `.partialUpdate(INKPLATE_FORCE_PARTIAL /*= true */)` to do the partial update.

This is necessary to populate the `_partial` buffer (defined [here](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/2dd263f2f9548aadac8638413a143beddf068a64/src/include/Graphics.h#L70)), which stores the "old" state that the partial update is done against.

Probably the easiest thing to do is to make a copy of `_partial` (or `DMemoryNew` if `_partial` is "private") and store it along with the counter value and whatever else using [ESPHome's "preferences" mechanism](https://github.com/esphome/esphome/blob/387bde665e3ae5d4b8d5a68299abd6032c4291b8/esphome/core/preferences.h).

ryrun, 2025-05-10:
I'm also in favor of adding this. I use an Inkplate on battery and would love to combine deep sleep with partial updates to save power and avoid full refreshes every time. Would definitely use this feature.