esphome/feature-requests #2208: Could Modbus Component support connect modbus tcp server?

645340633, 2023-04-17:
**Describe the problem you have/What new integration you would like**

Could Modbus Component support connect modbus tcp server? 
I have a rs485 modbus to tcp network server, but the esphome modbus component  only support uart module,so if it could support rs485 modbus to tcp network module or server?
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

makuser, 2023-10-05:
Are you asking for ESPHome to act as a Modbus TCP server for other Modbus TCP Clients to be able to connect to or are you asking for ESPHome to connect to another Modbus TCP server as a Modbus TCP client?

I understood your request to be able to connect to the Modbus TCP server you already seem to have, as a Modbus TCP client, which for obvious reasons can not be done over the only currently implemented physical protocol Modbus RTU by a RS485 Transceiver connected via UART.
If so, this is a duplicate of  #1992, but https://github.com/esphome/esphome/pull/4874 would do nothing for you. I have really no idea, why https://github.com/esphome/esphome/pull/4874 marked this FR as being solved by it...

If against all odds do you want ESPHome to act as a Modbus TCP Server and for example either talk Modbus RTU via UART or instead also provide your own registers via a lambda call or so, then https://github.com/esphome/esphome/pull/4874 would indeed implement this.
But in that case, this FR is a duplicate of #708.

Either way, for both Modbus TCP Server or Client, FRs already exist and this one is a duplicate.

645340633, 2023-10-05:
I mean if it could be like  modbus integration of homeassistant.  There are lots of differences between esphome  module and homeassistant integration. so I hope esphome cloud add the tcp protocol for the modbus module.

 

 
 If so, this is a duplicate of  #1992, but esphome/esphome#4874 would do nothing for you. I have really no idea, why esphome/esphome#4874 marked this FR as being solved by it...
 
If against all odds do you want ESPHome to act as a Modbus TCP Server and for example either talk Modbus RTU via UART or instead also provide your own registers via a lambda call or so, then esphome/esphome#4874 would indeed implement this.

makuser, 2023-10-05:
> I mean if it could be like&nbsp; modbus integration of homeassistant.&nbsp; There are lots of differences between esphome&nbsp; module and homeassistant integration. so I hope esphome cloud add the tcp protocol for the modbus module.

So you are looking for a Modbus TCP client functionality. Then I suggest following #1992.

This is a duplicate of #1992, please close this issue.

nagyrobi, 2023-10-06:
Duplicate of https://github.com/esphome/feature-requests/issues/1992

closed 2023-10-06 by nagyrobi