esphome/feature-requests #2210: Add support for LD2410 disable bluetooth

twocolors, 2023-04-21:
**Describe the problem you have/What new integration you would like**
Add support for LD2410 disable bluetooth

**Please describe your use case for this integration and alternatives you've tried:**

[image]

**Additional context**

@sebcaps

sebcaps, 2023-04-21:
Request handled by  https://github.com/esphome/esphome/pull/4434.
Need patience to be merged in official release, but can already be used , with esphome dev branch and
 `external_components:
    source: github://regevbr/esphome@ld2410
    components: [ld2410]
 `

twocolors, 2023-04-23:
oh, thx

closed 2023-04-23 by twocolors

ilvacca, 2023-06-09:
Hi guys. I would like to disable bluetooth on my LD2410 but I don't know how to use the commands you listed in the first comment. Can you explain me? Thank you very much.