esphome/feature-requests #2211: Add Support for Martin Jerry 3-Way Tasmota Switch (MJ-ST02)

JohnGalt1717, 2023-04-21:
**Describe the problem you have/What new integration you would like**
Currently this switch has Tasmota already on it, but flashing to ESPHome proves problematic because the relay switches immediately to off so there doesn't appear to be a way to talk to the secondary MCU that this device has and control the switch properly with state (you can just offer a toggle as far as I can tell)

Here's the link to it: https://www.amazon.com/dp/B0BJ9D1NM1

**Please describe your use case for this integration and alternatives you've tried:**
This is the instructions for Tasmota configuration:
```

1. Configure the module as 'Tuya MCU (54)'
2. Set the template to:

{"NAME":"MJ 3Way Switch","GPIO":[255,255,255,255,52,53,0,0,21,9,157,255,0],"FLAG":0,"BASE":18}

3. In the console, enter:

switchmode1 5
TuyaMCU 11,1
```

Which works fine in Tasmota with MQTT but I don't use MQTT.  I'd like to be able to create a configuration similar to the MJ-S01 that's in the device documentation and doesn't use Tuya, that sets the LEDs the same way depending on status, and reports proper light status in addition to being able to turn on and off the light.

**Additional context**
Seems to be secondary mcu stuff and I couldn't find any documentation to solve this myself to control communications to the secondary MCU or even how to monitor what the secondary MCU is sending back to the primary one so that I could fake it based on the values that it receives.

enablingpenguin, 2023-07-10:
This config worked for me

```
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600 #115200

tuya:

light:
  - platform: "tuya"
    name: ${friendly_name}
    switch_datapoint: 1
```

JohnGalt1717, 2023-07-10:
This works (and what I have) but it doesn't track the state of the switch. It's just a toggle.

enablingpenguin, 2023-07-10:
Weird mine tracks it, it even tracks if the dumb switch turns the light on or off.

I also have this set since I grabbed the config from ST01 on accident at first.

esp8266:
  # Using esp01 even though esptool shows 2MB flash
  board: esp01_1m

JohnGalt1717, 2023-07-10:
Ya, I have that.  The state when I visit the captive portal shows "off" even when the light is on which is reflected in Home Assistant.

carltonwb, 2023-10-07:
@JohnGalt1717 I just purchased this switch, without realizing that there was no esphome yaml prebuilt.
Would you mind sharing what you are currently using for the ST02 switch.
Thank you

kylhill, 2024-01-04:
This is the config I use: https://devices.esphome.io/devices/Martin-Jerry-MJ-ST02