esphome/feature-requests #2212: Modify ultrasonic sensor code to include the temperature for more precision

Majklzbastlirny, 2023-04-24:
**Describe the problem you have/What new integration you would like**

I would like to have temperature in ultrasonic distance sensor code for better precision. 
I have done a little research and in its code now the speed of sound is set to 343 m/s = 19.8°C. 
By default it would also be around 20°C but the arg would take the temperature (and maybe the format, default °C)
This is the formula for calculating the speed of sound in air:
[image]

Source for that: https://www.omnicalculator.com/physics/speed-of-sound#speed-of-sound-in-air

**Please describe your use case for this integration and alternatives you've tried:**

I'm building a device, which checks the water level of a water well. I want it to calculate: water height, volume of the well (in Liters and %), time to empty (so that water pump can shut down safely) and time to fill. 
Everything is based on the distance so if the reading is off , the true capacity will be off.

It is possibe to make it now, but only as filters: - lambda, but its messy. You have to extract the round-trip-time from the distance (with the hardcoded temperature of 19,8°C) and then multiply it by the real speed of sound, RTT and temperature.

**Additional context**

Example config:
sensor:
  - platform: ultrasonic
    trigger_pin: D1
    echo_pin: D2
    name: "Ultrasonic Sensor"
    update_interval: 1s
    temperature: 5 or _blank_ (defaults to 20°C) or id(tempsens).state (or something like it, directly from sensor)
    units: C (default), F, K

AndersWestman, 2023-07-31:
I agree that a temp parameter would be an elegant solution, but I doubt that there are that many who actually need that precision, except in those kind of level/volume measuring. 

With that said, I'm experience the same problem measuring the volume in a 1000l water tank.
As the speed change roughly 0,18% / °C (I think) it may be possible to do the compensation by just multiply 0,18% with the temp diff from 20 °C.

Something like

```
filters:
  - lambda: return x * ( ((id(tempsens).state - 20) * 0.0018) +1);
```

I have not test this, neither thought it through very much, so it's possible there is some brain fart here that I don't see... ;)

nanni85, 2023-09-06:
hello @Majklzbastlirny  you have solved?
can you share the config?