esphome/feature-requests #2214: key-collector compatibility for keypad feature of the sx1509 i/o expander

Meetsch, 2023-04-24:
**Describe the problem you have/What new integration you would like**
The [matrix-keypad](https://esphome.io/components/matrix_keypad.html) and [key-collector](https://esphome.io/components/key_collector.html) components works nicely together (thanks to @ssieb) , but the [keypad feature of the SX1509 I/O expander](https://esphome.io/components/sx1509.html?highlight=keypad#configuration-variables) is not compatible with the [key-collector](https://esphome.io/components/key_collector.html).

It would be great to make the SX1509 keypad implement the [key_provider class](https://github.com/esphome/esphome/blob/dev/esphome/components/key_provider/key_provider.h) to allow it to send keys and make it compatible with the key-collector natively.

**Please describe your use case for this integration and alternatives you've tried:**
I've been successfuly using the matrix-keypad to create an alarm panel, gathering the code through the key-collector.
But the matrix-keypad requires 8 pins to work (4x4 matrix).
As my project expands I'm now missing pins therefore I would like to connect the keypad through I2C, and this is the reason of trying to use the SX1509 I/O expander to interface with the matrix keypad and/or key-collector.

@ssieb : Is it already possible to use the sx1509 keypad with your key-collector? if yes could you please explain how? otherwise, would it be possible to make the sx1509 keypad implement the key_provider class?

**Additional context**
none

Meetsch, 2023-04-26:
> This FR could be applicable to TM1637, TM1638...?

No, I don't think so.

This feature request is for the use of [matrix keypads like these ones](https://learn.adafruit.com/matrix-keypad/overview)
[image]

Nothing related with 8 digit displays I find when I search for TM1637, TM1638.

nagyrobi, 2023-04-26:
TM1637 has a 2x8 key-matrix builtin...

https://esphome.io/components/display/tm1637.html#binary-sensor