esphome/feature-requests #2215: ESPHome enhancements for large homes

robsonfelix, 2023-04-24:
**Describe the problem you have/What new integration you would like**
Ability to handle large volume (100+) of ESPHome devices in Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**

ESPHome would greatly benefit from having capabilities to handle operations over multiple devices when the number of devices is big. I have a install with over 100 ESPhome devices.

It would be easier to handle such amount of devices by having a grid view (instead of an icon view), with additional information such as:

IP
version of ESPhome
etc
This grid view would enhance user experience if it allowed the content to be filtered by name, file contents, devices needing upgrade, offline devices, etc.

To top it all off, it would be nice to have the ability to perform operations in bulk, such as generate & download, install wirelessly (OTA) and validate.

I am also experiencing difficulty when using the “UPDATE ALL” capability. After running for half an hour, the browser gives up with a “out of memory exception”. I am running the browser on a notebook with a Core i9 with 32 GB of RAM (!!!).

I am also experiencing difficulty when opening multiple tabs of Home Assistant and doing the OTA update over wireless, as the compiling process is dragging itself out.

My Home Assistant install is running on a VM with 2 cores, 2 threads, and 8 GB of RAM. The underlying hardware is a brand new Dell R250 server with plenty of RAM and SSD storage.

Enhancements described above would greatly benefit any heavy user of ESPHome.

**Additional context**

fdcastel, 2023-06-18:
Related: #2205

robsonfelix, 2023-07-12:
Any ETA on this? I have a big list of devices that keeps growing and updating them is becoming extremely painful.