esphome/feature-requests #2219: Support Hi-Link LD2420 (HLK-LD2420) 24GHz Radar Sensor

zimmra, 2023-05-02:
**Describe the problem you have/What new integration you would like**

* Similar to the LD2410 Module, the LD2420 is a 24GHz Radar Sensor

**Please describe your use case for this integration and alternatives you've tried:**

* Use LD2420 in place of LD2410

**Additional context**

* Protocol information can be found at https://github.com/soubhik-khan/HLK-LD2420
* https://www.hlktech.net/index.php?id=1150

max246, 2023-05-13:
I will look into this

descipher, 2023-05-29:
Inprogress see: https://github.com/esphome/esphome/pull/4847

max246, 2023-05-29:
> Inprogress see: [esphome/esphome#4847](https://github.com/esphome/esphome/pull/4847)

To ensure it overrides my PR link, you would need to add this link to your PR description.

AlexKey1971, 2023-07-13:
Hello! Sometimes LD2420 freezes after firmware update on ESP32. It looks like "Moving Distance" always shows 35cm. Just restarting the ESP32 does not help. Only restarting the ESP32 on power helps. The code is standard, from here:
https://deploy-preview-2956--esphhome.netlify.app/components/sensor/ld2420

descipher, 2023-07-13:
It’s not something we can control, the ld2420 can be sent a reset command and this code does that but if it’s locked up then the only recourse is for the ld2420 is to use a watchdog function (unknown based on its docs) or a power cycle. This issue could be something else such as the baud gets changed or the unit is placed in a mode the does not transmit what we are expecting. The reset would be sent but not processed if the baud got changed. It possible to write code that walks the possible bauds and sends a reset on each but we do not know if this is the issue. You should see a command timeout error when the ld2420 is in an non-responsive state but it requires serial logging to observe.

descipher, 2023-07-13:
BTW that netlify link fails as 404.

AlexKey1971, 2023-07-13:
Corrected the link: https://deploy-preview-2956--esphome.netlify.app/components/sensor/ld2420

descipher, 2023-08-10:
I suspect there is an issue with the modules firmware. Please log the firmware version you have if you are observing hanging behaviour.

Mikescotland, 2023-08-10:
I have no idea how to check it, tried verbose mode and doesn't return the firmware version. However, I found the workaround which tested since yesterday and seems to resolve the problem.
I added the restart switch in the config (platform restart). Once after ota the module locks, restarting esp from restart platform solves the issue. I don't know if esp restart after ota is different than restart via platform, however, haven't had any problems with it since. BTW my module works only with external component github://pr#4847, the one stated in the esphome website with config yaml, I couldn't make it work at all.

descipher, 2023-08-10:
The version is logged at setup. Should see it with HA wireless api logging. Otherwise it would need a serial connection on the logging port. I can add it to the web interface as well.

descipher, 2023-08-10:
The only code available that works is the PR #4847 that I submitted, it has yet to be reviewed so it is not in any release form yet.

descipher, 2023-08-10:
Using HA's ESPhome control panel integration ..

[image]

[image]

Logs -> Wireless

[image]

Mikescotland, 2023-08-10:
How did I miss it? I mean not the way to retrieve it, I just simply missed this value when reviewing.
Version 1.5.6

descipher, 2023-08-10:
The OTA failure issue does not look like a modual lockup because it corrects after an ESP restart. When I observe the lockup it will never correct with any ESP restart. It persists until power is cycled. Thanks for the version info. I believe Hi-Link will be providing firmware update capability via serial and it may already be available in v1.5.6. We need to wait for them to publish the information.

hkrob, 2023-08-18:
Same behaviour here..

```
[10:59:57][C][logger:301]: Logger:
[10:59:57][C][logger:302]:   Level: DEBUG
[10:59:57][C][logger:303]:   Log Baud Rate: 115200
[10:59:57][C][logger:305]:   Hardware UART: UART0
[10:59:57][C][uart.arduino_esp32:124]: UART Bus 1:
[10:59:57][C][uart.arduino_esp32:125]:   TX Pin: GPIO32
[10:59:57][C][uart.arduino_esp32:126]:   RX Pin: GPIO33
[10:59:57][C][uart.arduino_esp32:128]:   RX Buffer Size: 256
[10:59:57][C][uart.arduino_esp32:130]:   Baud Rate: 256000 baud
[10:59:57][C][uart.arduino_esp32:131]:   Data Bits: 8
[10:59:57][C][uart.arduino_esp32:132]:   Parity: NONE
[10:59:57][C][uart.arduino_esp32:133]:   Stop bits: 1
[10:59:57][C][ld2420:012]: LD2420:
[10:59:57][C][ld2420:013]:   Firmware Version :  v1.5.1
```

hkrob, 2023-09-02:
Anyone know where to find the updated firmware?

lboue, 2023-10-15:
Anyone know where to find the updated firmware?

descipher, 2023-10-20:
> Anyone know where to find the updated firmware?

You have to email them to get them.
I have found that only v1.5.6 and up work. If you try to update v1.5.4 you will brick it!
v1.5.3 and older cannot be updated. They would need to be done via JTAG SW .. event with that  .. not for the inexperienced (its complex to say the least).

gkawanami, 2023-11-05:
Hi to all! 

Problem: Cannot install ESPHome on AtomLite (ESP32) board with LD2420

Set-UP:
ESP-Home 2023.10.6
HA 2023.10.3
ESP32 = Atom Lite [https://docs.m5stack.com/en/core/atom_lite#Description](url)
Sensor = LD2420 [https://deploy-preview-2956--esphome.netlify.app/components/sensor/ld2420](url)

Description:
When installing this yaml file (code below), file preparation shows the error:

> 
INFO ESPHome 2023.10.6
INFO Reading configuration /config/esphome/presence01.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/4847/head
Failed config
> 

Question:
Am I using the wrong external component link?
What am I doing wrong?

YAML File:
`external_components:
  - source: github://pr#4847
    components: [ ld2420 ]
    refresh: 0s

esphome:
  name: "mmwave01"
  friendly_name: "mmwave01"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "cTy9RiVi61wyb1dnvAQqiPMNTyl5jviJO4lQ/aC7jXc="

ota:
  password: "20755a16743d40176874c4f8625013a0"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Teste Fallback Hotspot"
    password: "mtem1GYyg9YP"

captive_portal:

esp32_ble_tracker:
    
uart:
  id: ld2420_uart
  tx_pin: 22
  rx_pin: 19
  baud_rate: 115200
  parity: NONE
  stop_bits: 1
[... 177 lines cut ...]
    g11_still_threshold: 100
    
    [g12_move_threshold] is an invalid option for [0]. Please check the indentation.
    g12_move_threshold: 250
    
    [g12_still_threshold] is an invalid option for [0]. Please check the indentation.
    g12_still_threshold: 100
    
    [g13_move_threshold] is an invalid option for [0]. Please check the indentation.
    g13_move_threshold: 200
    
    [g13_still_threshold] is an invalid option for [0]. Please check the indentation.
    g13_still_threshold: 100
    
    [g14_move_threshold] is an invalid option for [0]. Please check the indentation.
    g14_move_threshold: 200
    
    [g14_still_threshold] is an invalid option for [0]. Please check the indentation.
    g14_still_threshold: 100
    
    [g15_move_threshold] is an invalid option for [0]. Please check the indentation.
    g15_move_threshold: 200
    
    [g15_still_threshold] is an invalid option for [0]. Please check the indentation.
    g15_still_threshold: 100`

descipher, 2023-11-05:
@gkawanami There are new features. Check the YAML at the beginning of this PR and the doc link.

dhnsilva, 2023-11-06:
@descipher using esp8266 
Using de code until:

```
 text_sensor:
   - platform: ld2420
     fw_version:
       name: LD2420 Firmware
 
 sensor:
   - platform: ld2420
     moving_distance:
       name : Moving Distance
 
 binary_sensor:
   - platform: ld2420
     has_target:
       name: Presence
```

I got this problem:

```
INFO Reading configuration /config/radar-01.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/4847/head
INFO Generating C++ source...
INFO Compiling app...
Processing teste (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/teste/src/esphome/components/ld2420/ld2420.cpp.o
Compiling .pioenvs/teste/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/teste/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/teste/src/esphome/components/mdns/mdns_component.cpp.o
src/esphome/components/ld2420/ld2420.cpp: In member function 'virtual void esphome::ld2420::LD2420Component::dump_config()':
src/esphome/components/ld2420/ld2420.cpp:71:44: error: 'class esphome::ld2420::LD2420Component' has no member named 'gate_timeout_number_'
   71 |   LOG_NUMBER(TAG, "  Gate Timeout:", this->gate_timeout_number_);
      |                                            ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:71:3: error: 'LOG_NUMBER' was not declared in this scope
   71 |   LOG_NUMBER(TAG, "  Gate Timeout:", this->gate_timeout_number_);
      |   ^~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:72:49: error: 'class esphome::ld2420::LD2420Component' has no member named 'max_gate_distance_number_'
   72 |   LOG_NUMBER(TAG, "  Gate Max Distance:", this->max_gate_distance_number_);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:73:49: error: 'class esphome::ld2420::LD2420Component' has no member named 'min_gate_distance_number_'
   73 |   LOG_NUMBER(TAG, "  Gate Min Distance:", this->min_gate_distance_number_);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:74:43: error: 'class esphome::ld2420::LD2420Component' has no member named 'gate_select_number_'
   74 |   LOG_NUMBER(TAG, "  Gate Select:", this->gate_select_number_);
      |                                           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:76:53: error: 'class esphome::ld2420::LD2420Component' has no member named 'gate_move_threshold_numbers_'
   76 |     LOG_NUMBER(TAG, "  Gate Move Threshold:", this->gate_move_threshold_numbers_[gate]);
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:77:55: error: 'class esphome::ld2420::LD2420Component' has no member named 'gate_still_threshold_numbers_'
   77 |     LOG_NUMBER(TAG, "  Gate Still Threshold::", this->gate_still_threshold_numbers_[gate]);
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:79:44: error: 'class esphome::ld2420::LD2420Component' has no member named 'apply_config_button_'; did you mean 'apply_config_action'?
   79 |   LOG_BUTTON(TAG, "  Apply Config:", this->apply_config_button_);
[... 32 lines cut ...]
src/esphome/components/ld2420/ld2420.cpp:186:9: error: 'class esphome::ld2420::LD2420Component' has no member named 'gate_timeout_number_'
  186 |   this->gate_timeout_number_->state = FACTORY_TIMEOUT;
      |         ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:187:9: error: 'class esphome::ld2420::LD2420Component' has no member named 'min_gate_distance_number_'
  187 |   this->min_gate_distance_number_->state = FACTORY_MIN_GATE;
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:188:9: error: 'class esphome::ld2420::LD2420Component' has no member named 'max_gate_distance_number_'
  188 |   this->max_gate_distance_number_->state = FACTORY_MAX_GATE;
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp: In member function 'void esphome::ld2420::LD2420Component::set_operating_mode(const string&)':
src/esphome/components/ld2420/ld2420.cpp:289:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
  289 |     this->operating_selector_->publish_state(state);
      |           ^~~~~~~~~~~~~~~~~~~
src/esphome/components/ld2420/ld2420.cpp:308:11: error: 'class esphome::ld2420::LD2420Component' has no member named 'operating_selector_'
  308 |     this->operating_selector_->publish_state(OP_SIMPLE_MODE_STRING);
      |           ^~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/teste/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/teste/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/teste/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/teste/src/esphome/components/network/util.cpp.o
*** [.pioenvs/teste/src/esphome/components/ld2420/ld2420.cpp.o] Error 1
========================= [FAILED] Took 22.16 seconds =========================
```

If I put complete:

descipher, 2023-11-06:
@dhnsilva you need to add the following for now.

~~~
number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode
~~~

dhnsilva, 2023-11-07:
> @dhnsilva you need to add the following for now.
>
> ```
> number:
>   - platform: ld2420
>     presence_timeout:
>       name: Detection Presence Timeout
>     min_gate_distance:
>       name: Detection Gate [...]

adding this I got this error:

[image]

descipher, 2023-11-08:
This will compile - 
~~~
# The LD2420 has 16 sense gates 0-15 and each gate detects 0.7 meters 15th gate = 9m
ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

sensor:
  - platform: ld2420
    moving_distance:
      name : Moving Distance

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

switch:
  - platform: restart
    name: Restart ESPHome Node

select:
  - platform: ld2420
    operating_mode:
      name: (s) Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: (n) Detection Presence Timeout
    min_gate_distance:
      name: (n) Detection Gate Minimum
    max_gate_distance:
      name: (n) Detection Gate Maximum
    gate_select:
      name: (n) Select Gate to Set
    still_threshold:
      name: (n) Set Still Threshold Value
    move_threshold:
      name: (n) Set Move Threshold Value

button:
  - platform: ld2420
    apply_config:
      name: (b) Apply Config
    factory_reset:
      name: (b) Factory Reset
    restart_module:
      name: (b) Restart Module
    revert_config:
      name: (b) Undo Edits
~~~

dhnsilva, 2023-11-08:
It Is incredible!

when I put only text_sensor, sensor, binary_sensor, switch,  like this:

[image]

It starts to compiling, but obviously don't finishes.

[image]

If I put the complete code like yours (With select, number and button), the compile appears with a a error on the beginning, like this....

[image]

I have 5 ESP8622 and 5 LD2420 and I would like to use it so much!

descipher, 2023-11-08:
This part is required.

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

dhnsilva, 2023-11-08:
yes... is with select entry also! 
Identic with

```
select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode
```

Or 

```
select:
  - platform: ld2420
    operating_mode:
      name: (s) Operating Mode
```

The error is the same!
[image]

descipher, 2023-11-08:
@dhnsilva Ok, I can test an ESP8266 what specific device do you have defined in the YAML.

descipher, 2023-11-08:
@dhnsilva It does compile for me using an ESP8266, What version of ESPHome are you running?

dhnsilva, 2023-11-09:
thank you very much!
I was almost giving up!
ESPHome version is 2023.2.4

dhnsilva, 2023-11-09:
Other info: 
I am tried 
```
select:
  - platform: ld2420
    operating_mode:
      name: (s) Operating Mode
```
and

```
select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode
```

descipher, 2023-11-09:
@dhnsilva ESPHome version is 2023.2.4, that's 8 months old. Try updating.

We are currently running:
INFO ESPHome 2023.10.6 
INFO Reading configuration /config/esphome/test.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/4847/head

Is this an older ld2420? with fw v1.5.3 and up the baud rate is 115200

Also that module will not work with logging at the same time. 
either remove logger
or disable logging

dhnsilva, 2023-11-10:
@descipher 
It works!!!!!!
the problem was the version of ESPHome as you said!

Iam using a docker version of EspHome and the link used was out of date.
The update was enough to get ir working!
One more time!
Thanks!

descipher, 2023-11-10:
Awesome .. thanks for the update. There are some things I need to edit in the code. Specifically I missed some #ifdef USE_NUMBER and the USE_TEXT .. normally you will be able to use just the parts you need outside of the required select:

vb7445, 2023-12-31:
ESPHome 2023.12.5
Home Assistant OS,  core-2023.12.4
LD2420, version 1.5.9

I'm using the standard yaml for the LD2420. The functionality seems to be there, i.e. Calibration works sending data to the log. However, I'm not getting proper detections. In other words, it's not detecting movement. I've tried various thresholds, calibrations and factory resets. I tried a couple different devices but got the same results.
I'm wondering if there's some trick to setting this up or should it run right out of the box so to say?`

```
esphome:
  name: "ehproto"
  friendly_name: ehproto

esp8266:
  board: esp01_1m

# Logging serial output disabled to use LD2420, baud_rate: 0
logger:
   level: DEBUG
   baud_rate: 0

   
# Enable Home Assistant API
api:
  encryption:
    key: "grcIxcAUbEMUVVk5z2DbeUAh8BUdkJRFgoVLX62Eu0I="

ota:

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
      channel: 11
  manual_ip: 
    static_ip: 192.168.1.60
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  power_save_mode: none
  output_power: 20dB
  fast_connect: True
  reboot_timeout: 5min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-Aad0Ea"
    password: "dBuI0myRwuG5"

captive_portal:

uart:
  id: ld2420_uart
  tx_pin: 1
  rx_pin: 3
  baud_rate:  115200
  # parity: NONE
  # stop_bits: 1

# The LD2420 has 16 sense gates 0-15 and each gate detects 0.7 meters 15th gate = 9m
ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
      name: Set Move Threshold Value

button:
  - platform: ld2420
    apply_config:
      name: Apply Config
    factory_reset:
      name: Factory Reset
    restart_module:
      name: Restart Module
    revert_config:
      name: Undo Edits

sensor:
  - platform: adc
    pin: A0
    name: "Battery Voltage"
    update_interval: 3min
    accuracy_decimals: 2
    filters:
      - multiply: 1
      - calibrate_linear:
         method: least_squares
         datapoints:
          # Map 0.0 (from sensor) to 1.0 (true value)
          - 0.0 -> 0.0
          - 0.83 -> 3.87
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 3min
  - platform: ld2420
    moving_distance:
      name : Moving Distance
```

descipher, 2024-01-01:
@vb7445 Does it report the firmware correctly? Is the device in an enclosure?

vb7445, 2024-01-01:
@descipher  I don't have it in an enclosure but open on the bench. The proto board it's on is being held in a vise which is pointing the LD2420 out into the room as it's detailed in the documentation for wall mounting. Yes, it does report the firmware as v1.5.9. I see that in Home Assistant based on the text sensor in the config file:

```yaml
text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware
```

[image]

descipher, 2024-01-01:
Since your actually using a D1 mini your config could be changed.

Don't use the Tx Rx pins for the uart, Use the following edits.

~~~

logger:

esp8266:
  board: d1_mini

uart:
  id: ld2420_uart
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 115200
  parity: NONE
  stop_bits: 1
~~~

vb7445, 2024-01-02:
@descipher  Very good, those settings work. Thanks.
I connected the LD2420 OT1 pin3 to GPIO4 and LD2420 RX pin4 to GPIO5. 
Also had to run a Calibration as the Factory settings didn't work for my setup.

I was of the understanding that the I2C bus on the D1 use pins GPIO4 & 5 which I was going to use with a LHT20 sensor, so I'll need to better understand these GPIO pins on a D1. I admit I get confused with ESP device pinouts together with DEV Kit pinouts and what can be used with what.

descipher, 2024-01-02:
@vb7445 You're welcome. Yes those are the I2C hardware pins, I would suggest using an esp32 D1 board which have advanced peripheral pin mapping capability for the use case you're dialoging. You will be glad you have the extra memory with an esp32 vs a esp8266.

miguellandre, 2024-01-14:
I'm using the default yaml for the LD2420. However, I am not getting detections, it appears unknown. I have tried factory resets. I tried a few different codes, but always the same results.
Is there anything else I can do to configure it?
I also leave the logs to see if there is anything wrong.

my code add logs:

`
esphome:
  name: ld2024teste
  friendly_name: ld2024teste

esp8266:
  board: esp01_1m

# Logging serial output disabled to use LD2420, baud_rate: 0
logger:
   level: DEBUG
   baud_rate: 0
# Enable Home Assistant API
api:
  encryption:
    key: "Hq/6Kpni97Izlk+86/7lpSiXMwLQo9kZbx5NOnk25Vo="

ota:
  password: "56a07268de888f5188cdf357943d44cc"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  id: ld2420_uart
  tx_pin: 1 
  rx_pin: 3
  baud_rate: 115200
  parity: NONE
  stop_bits: 1

ld2420:

text_sensor:
  - platform: ld2420
    fw_version:
      name: LD2420 Firmware

binary_sensor:
  - platform: ld2420
    has_target:
      name: Presence

select:
  - platform: ld2420
    operating_mode:
      name: Operating Mode

number:
  - platform: ld2420
    presence_timeout:
      name: Detection Presence Timeout
    min_gate_distance:
      name: Detection Gate Minimum
    max_gate_distance:
      name: Detection Gate Maximum
    gate_select:
      name: Select Gate to Set
    still_threshold:
      name: Set Still Threshold Value
    move_threshold:
[... 77 lines cut ...]
[14:50:35][C][ld2420:085]: ld2420  Icon: 'mdi:database'
[14:50:35][C][ld2420:087]: LD2420 Select:
[14:50:35][C][ld2420:088]: ld2420  Operating Mode 'Operating Mode'
[14:50:35][W][ld2420:090]: LD2420 Firmware Version v0.0.0 and older are only supported in Simple Mode
[14:50:35][C][LD2420.text_sensor:011]: LD2420 TextSensor:
[14:50:35][C][LD2420.text_sensor:012]:   Firmware 'LD2420 Firmware'
[14:50:35][C][LD2420.text_sensor:012]:     Icon: 'mdi:chip'
[14:50:35][C][LD2420.binary_sensor:011]: LD2420 BinarySensor:
[14:50:35][C][LD2420.binary_sensor:012]:   Presence 'Presence'
[14:50:35][C][LD2420.binary_sensor:012]:     Device Class: 'occupancy'
[14:50:35][C][LD2420.sensor:011]: LD2420 Sensor:
[14:50:35][C][LD2420.sensor:012]:   Distance 'Moving Distance'
[14:50:35][C][LD2420.sensor:012]:     Device Class: 'distance'
[14:50:35][C][LD2420.sensor:012]:     State Class: ''
[14:50:35][C][LD2420.sensor:012]:     Unit of Measurement: 'cm'
[14:50:35][C][LD2420.sensor:012]:     Accuracy Decimals: 0
[14:50:35][C][captive_portal:088]: Captive Portal:
[14:50:35][C][mdns:115]: mDNS:
[14:50:35][C][mdns:116]:   Hostname: ld2024teste
[14:50:35][C][ota:097]: Over-The-Air Updates:
[14:50:35][C][ota:098]:   Address: ld2024teste.local:8266
[14:50:35][C][ota:101]:   Using Password.
[14:50:35][C][api:139]: API Server:
[14:50:35][C][api:140]:   Address: ld2024teste.local:6053
[14:50:35][C][api:142]:   Using noise encryption: YES

descipher, 2024-01-15:
@miguellandre There are a number of things you need to rule out.

1) Is the module running at 256000 or 115200 early modules were 256000 e.g. v1.5.2 or lower.
2) Are the serial lines connected correctly? Don't assume someone else's config works as is.
LD Tx to ESP Rx
LD Rx to ESP Tx
[image]
If the firmware is unknown then do all the possible firmware pin maps. 
3) Simplify the config to only the required elements.
4) esp_01m have variations in capability is yours capable?
5) Logging? Try removing the logging component all together.

Huymee, 2024-01-19:
> I'm using the default yaml for the LD2420. However, I am not getting detections, it appears unknown. I have tried factory resets. I tried a few different codes, but always the same results. Is there [...]

U can try this:
```
uart:
  #id: ld2420_uart  
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 115200
  parity: NONE
  stop_bits: 1
```

descipher, 2024-01-19:
If it's an actual ESP01 module those pins are not accessible without precision soldering and magnet wire, only 1 and 3 can be accessed. The ESP01 is not recommended for this reason.

Huymee, 2024-01-19:
> If it's an actual ESP01 module those pins are not accessible without precision soldering and magnet wire, only 1 and 3 can be accessed. The ESP01 is not recommended for this reason.

It's works on ESP01 with GPIO1 & GPIO3 but LD2420 needs to be powered on first before ESP01.
Other ESP are also.

itestoni, 2024-01-20:
> Same here. Works flawlessly on Esp32, however becomes "frozen" if OTA applied or soft restart of Esp32. Only power cycle helps. Shame, as this sensor is somehow better behaving in my environment than the Ld2410.

Same as me.
I have d1-mini esp8266 connected.
I don't know if the issue is after reboot but sometimes it gets frozen reporting always 35cm.
My FW is 1.5.9 and default configuration.
I use it with ESPHome.
I suspect could be a power supply issue.
To the same power supply (5V/700mA) I have connected also another module with esp32.
Possibly there could be some spike/drop on the voltage.
I have added on LD2420 3.3V/gnd input a 100uF electrolytic cap (the one I had available at home).
Let's see if it works.

Huymee, 2024-01-20:
当你使用ESP01的时候可以配置以下信息，将日志串口切换到别的pin，然后就可是正确输出LD2420的固件版本了
```
# Enable logging
logger:
  hardware_uart: UART1
```

itestoni, 2024-01-20:
> 当你使用ESP01的时候可以配置以下信息，将日志串口切换到别的pin，然后就可是正确输出LD2420的固件版本了
> 
> ```
> # Enable logging
> logger:
>   hardware_uart: UART1
> ```

Hi, FW version is 1.5.9, here following ESPHome log.
Anyway, putting a capacitor on 3v3 didn't help.

[16:51:24][C][logger:439]: Logger:
[16:51:24][C][logger:440]:   Level: DEBUG
[16:51:24][C][logger:441]:   Log Baud Rate: 115200
[16:51:24][C][logger:443]:   Hardware UART: UART0
[16:51:24][C][uart.arduino_esp8266:102]: UART Bus:
[16:51:24][C][uart.arduino_esp8266:103]:   TX Pin: GPIO4
[16:51:24][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[16:51:24][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[16:51:24][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[16:51:24][C][uart.arduino_esp8266:109]:   Data Bits: 8
[16:51:24][C][uart.arduino_esp8266:110]:   Parity: NONE
[16:51:24][C][uart.arduino_esp8266:111]:   Stop bits: 1
[16:51:24][C][uart.arduino_esp8266:115]:   Using software serial
[16:51:24][C][ld2420:068]: LD2420:
[16:51:24][C][ld2420:069]:   Firmware Version :  v1.5.9
[16:51:24][C][ld2420:070]: LD2420 Number:
[16:51:24][C][ld2420:072]: ld2420  Gate Timeout: 'Timeout di rilevazione presenza'
[16:51:24][C][ld2420:072]: ld2420  Icon: 'mdi:timelapse'
[16:51:24][C][ld2420:072]: ld2420  Unit of Measurement: 's'
[16:51:24][C][ld2420:073]: ld2420  Gate Max Distance: 'Distanza massima gate'
[16:51:24][C][ld2420:073]: ld2420  Icon: 'mdi:motion-sensor'
[16:51:24][C][ld2420:073]: ld2420  Device Class: 'distance'
[16:51:24][C][ld2420:074]: ld2420  Gate Min Distance: 'Distanza minima gate'

Huymee, 2024-01-20:
> Hi, FW version is 1.5.9, here following ESPHome log. Anyway, putting a capacitor on 3v3 didn't help.
>
> [16:51:24][C][logger:439]: Logger: [16:51:24][C][logger:440]: Level: DEBUG [...]

我的LD2420固件版本也是1.5.9
上面这个日志单独看起来没什么问题，不过我没有发现距离的报文。

另外你需要注意一下LD2420附近是不是有什么干扰源，我昨天测试的时候发现如果存在干扰的时候会导致雷达只上报10~12cm的数据(不同环境可能数值会不同)，后来我把雷达换了一个位置就不存在这个现象了。
另外还有一种情况是日志中出现“Firmware Version :v0.0.0”，这时候不会出现距离的报文，出现这种情况的时候给ESP模块单独断电(雷达不断电)一次之后，等待ESP模块重新启动就好了。 -----出现这种情况的原因我怀疑是ESP模块在初始化的时候LD2420还没有完成启动，然后ESP只向雷达发送一次初始化请求，但由于LD2420还未启动完成导致它没有收到初始化信息而不会通过串口向ESP发送数据，对于硬件这块我不是太了解，所以需要一位这方面的专家帮忙确认一下我这个猜想是否能帮忙解决这个问题。

itestoni, 2024-01-20:
I think I was able to replicate the issue.
To do it simply reduce the max distance to 5.
Then the distance will be always 35cm more or less...

Here the log when changing from 4 to 5 the max gate distance.

[17:24:23][D][number:012]: 'Distanza minima gate': Sending state 1.000000
[17:24:23][D][number:012]: 'Distanza massima gate': Sending state 4.000000
[17:24:23][D][number:012]: 'Impostazione soglia immobilità': Sending state 40000.000000
[17:24:23][D][number:012]: 'Impostazione soglia movimento': Sending state 60000.000000
[17:24:23][D][ld2420:671]: Sending write system mode command: 12
[17:24:23][D][ld2420:483]: LD2420 reply - set system parameter(s): 12 success
[17:24:23][D][ld2420:567]: Sending set config disable command: FE
[17:24:23][D][ld2420:450]: LD2420 reply - set config disable: CMD = FE success
[17:24:23][D][select:015]: 'Modalità operativa': Sending state Normal (index 0)
[17:24:23][C][ld2420:179]: LD2420 reconfig complete.
[17:24:23][W][component:214]: Component web_server took a long time for an operation (0.92 s).
[17:24:23][W][component:215]: Components should block for at most 20-30ms.
[17:24:24][D][sensor:093]: 'Distanza di movimento': Sending state 86.00000 cm with 0 decimals of accuracy
[17:24:27][D][sensor:093]: 'Distanza di movimento': Sending state 51.00000 cm with 0 decimals of accuracy
[17:24:28][D][sensor:093]: 'Distanza di movimento': Sending state 67.00000 cm with 0 decimals of accuracy
[17:24:30][D][sensor:093]: 'Distanza di movimento': Sending state 71.00000 cm with 0 decimals of accuracy
[17:24:31][D][sensor:093]: 'Distanza di movimento': Sending state 57.00000 cm with 0 decimals of accuracy
[17:24:32][D][sensor:093]: 'Distanza di movimento': Sending state 49.00000 cm with 0 decimals of accuracy
[17:24:33][D][sensor:093]: 'Distanza di movimento': Sending state 82.00000 cm with 0 decimals of accuracy
[17:24:34][D][sensor:093]: 'Distanza di movimento': Sending state 91.00000 cm with 0 decimals of accuracy
[17:24:35][D][sensor:093]: 'Distanza di movimento': Sending state 72.00000 cm with 0 decimals of accuracy
[17:24:37][D][sensor:093]: 'Distanza di movimento': Sending state 74.00000 cm with 0 decimals of accuracy
[17:24:38][D][sensor:093]: 'Distanza di movimento': Sending state 75.00000 cm with 0 decimals of accuracy
[17:24:39][D][sensor:093]: 'Distanza di movimento': Sending state 66.00000 cm with 0 decimals of accuracy
[17:24:45][D][number:054]: 'Distanza massima gate' - Setting number value
[17:24:45][D][number:113]:   New number value: 5.000000
[17:24:45][D][number:012]: 'Distanza massima gate': Sending state 5.000000
[17:24:48][D][button:010]: 'Applica configurazione' Pressed.
[17:24:48][C][ld2420:161]: Reconfiguring LD2420...
[17:24:48][D][ld2420:567]: Sending set config enable command: FF
[17:24:48][D][ld2420:447]: LD2420 reply - set config enable: CMD = FF success
[17:24:48][D][ld2420:716]: Sending write gate min max and timeout command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    0 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    1 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    2 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    3 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    4 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    5 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    6 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    7 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    8 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    9 sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    A sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    B sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    C sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    D sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    E sensitivity command:  7
[17:24:48][D][ld2420:468]: LD2420 reply - write gate parameter(s):  7 success
[17:24:48][D][ld2420:742]: Sending set gate    F sensitivity command:  7
[... 2 lines cut ...]
[17:24:48][D][number:012]: 'Selezione gate': Sending state 0.000000
[... cut ...]
[17:25:03][D][sensor:093]: 'Distanza di movimento': Sending state 40.00000 cm with 0 decimals of accuracy
[17:25:05][D][sensor:093]: 'Distanza di movimento': Sending state 35.00000 cm with 0 decimals of accuracy
[17:25:09][D][sensor:093]: 'Distanza di movimento': Sending state 40.00000 cm with 0 decimals of accuracy
[17:25:10][D][sensor:093]: 'Distanza di movimento': Sending state 35.00000 cm with 0 decimals of accuracy
[17:25:11][D][sensor:093]: 'Distanza di movimento': Sending state 41.00000 cm with 0 decimals of accuracy

Huymee, 2024-01-20:
是的，在使用之前需要在实际使用的环境中使用HLK官方提供的工具调试一下，获得门限数据之后才能正常使用，直接使用默认数据可能得不到你想要的结果

descipher, 2024-01-20:
The issue you are describing is known by HLK. The module will go into a state where all threshold register values are ignored and the detections are spurious. Used to be more frequent at 256000 baud. A power cycle will clear it. I can can create the issue by flooding the unit with a high rate of serial commands. Once configured it will be stable all the time. It's just when it has to process rapid register updates. I will experiment with a throttle function as a work around for now.

itestoni, 2024-01-20:
The point is that when in this condition I'm not able to revert it back.
It is stuck.
I need to cycle the power supply. :(

descipher, 2024-01-20:
Yes, I am frustrated by that, It's no a code issue. But I can probably work around it.

itestoni, 2024-01-20:
> Yes, I am frustrated by that, It's not a component code issue. I can probably work around it.

Ok but I got frozen the module also with no other apparent reason.
It was simply left on his own mounted on the wall overnight.
Then I found it stuck...
So, at least I have two different conditions to get it stuck.
By the way, I have no issue after OTA FW update like user before.

Huymee, 2024-01-20:
我刚才查了一下协议文档，可以发送以下命令重启LD2420：
```
FD FC FB FA 02 00 68 00 04 03 02 01
```

descipher, 2024-01-20:
With clean sufficient power I have never seen the module lockup. 

> I just checked the protocol document and I can send the following command to restart LD2420:
> 
> ```
> FD FC FB FA 02 00 68 00 04 03 02 01
> ```

The component code sends that command with the restart option button.

[image]

descipher, 2024-01-20:
It will not fix the problem since the module ignores all commands when this state.

hagak, 2024-01-22:
One issue I have noticed with on ld2420 v1.5.8 and esp32s3 is that it would not reliably connect to the ld2420 at setup.  I suspect it was because the LD2420 had not initialized before the esp32 attempted to connect to it and then the LD2420 was in the "locked state". This behaviour appeared to happen rather often at power up.

I was able to workaround this issue and it has been reliable at startup by adding this:
```
ld2420:
  setup_priority: -100
```
To prevent the esp32 from trying to setup the ld2420 too early and seems to be reliable now.

descipher, 2024-01-22:
@hagak Thanks,I will submit a PR to adjust the priority.

[78 more replies]

closed 2025-04-10 by zimmra