esphome/feature-requests #2221: ESP32 internal_temperature

anzejarni, 2023-05-02:
**Describe the problem you have/What new integration you would like**

Please add an optional 'difference' boolean, that enables the following function:
1. Read the the initial temperature on boot
2. Return the temperature difference between the read temperature and initial temperature

This would be very beneficial to be able to tell if the chip is overheating

**Please describe your use case for this integration and alternatives you've tried:**

I have several operational S2 ESP32 chips. Some reporting the right temperature and others are far off. They all however do have the same increase curve when they start heating up after boot. So the initial read temperature is wrong, however the change seems to be accurate.

**Additional context**

anzejarni, 2023-05-02:
[image]

Unfocused, 2023-05-03:
As an alternative, this could be done using a custom filter:

```yaml
sensor: 
  - platform: internal_temperature 
    name: "Internal temperature delta"
    filters: 
      - lambda: !lambda |- 
          // Using static will make this retain it's value between calls
          static float boot_temp = -1;
          if (boot_temp == -1) {
            // Store the value of only the first time
            boot_temp = value;
          }
          // Return the difference between the current temperature and the first temperature after boot 
          return (value - boot_temp);     
```
(Untested & written on my phone, so please excuse any typos & indentation issues)

anzejarni, 2023-05-03:
@Unfocused thanks!
Changed your code a bit to work.
```
sensor: 
  - platform: internal_temperature 
    name: "Internal temperature delta"
    filters: 
      - lambda: !lambda |- 
          // Using static will make this retain it's value between calls
          static float boot_temp = -999;
          if (boot_temp == -999) {
            // Store the value (x) of only the first time
            boot_temp = x;
          }
          // Return the difference between the current temperature and the first temperature after boot 
          return (x - boot_temp);  
```

But I still think it could be nice to have it integrated into the component if the codeowner has time.

Unfocused, 2023-05-06:
It could be added as a built-in filter, then it would be availble to all sensors