esphome/feature-requests #2223: [Feature Request] for Cloudflared tunnel support

deanfourie1, 2023-05-04:
Would love to see integration for the ESPhome project and ESP32 to support the possiblity to connect to a cloudflared tunnel. Not sure how possible this is to achieve, but I have a use case scenario where I want to use a TTGO T-Call to connect and send MQTT over WAN, but cannot open port due to CGNat.

Allowing the ESP and ESPhome to connect the ESP to a cloudflared tunnel would be pretty awesome!

PablexXXXX, 2023-05-16:
This would be amazing. Imagine if you could have an URL entry in your YAML pointing to your external home assistance instance. 
PLEASE DO THIS.

PablexXXXX, 2023-05-16:
ESPHome Cloudflare tunnel.

Ralphy140, 2023-05-23:
As described in a similar request this the challenge is that the ESPHome device is the server not the HA so the ESPHome device would need to setup the tunnel not the other way around. to be able to expose the ESPHome device you would need it to run the cloudflared tunnel code on the ESP and then create a new DNS endpoint. You would then need to point your home assistant server at the new end point.1) I am not sure that the ESP32 is powerfull enough to run that code and 2) A solution for remote and encrypted communication already exists, if you want to do remote ESP home devices it would be best to use encrypted/authenticated  MQTT that is pointed at a public MQTT broker then connect your HA instance to liten to that MQTT server, this still solve the "anywhere" challenge of deploying an ESPHome device with components that are already in place.

deanfourie1, 2023-05-23:
Yes but in order to setup remote MQTT you will need to have MQTT port open server side.

This is not possible for those behind CGNat.

The idea of the esp32 creating the tunnel is seemless, 

It executes the connection and creates the tunnel, 

The the cloudflare routing handles the rest, routing between other tunnels etc to reach the HA instance.

This is what would make this integration so awesome.

Ralphy140, 2023-05-23:
In this architecture you would use a public MQTT broker like hivemq, You would connect both the endpoint of the ESP and the HA server to it. Then it will do MQTT message exchange without ether device being public. I am not proposing that you make your HA server public

nagyrobi, 2023-05-23:
Cloudflared is "public" too... They do see inside your tunnels...

dougiteixeira, 2023-06-18:
I manage to expose my MQTT broker through the cloudflared tunnel, but only MQTT over WebSocket, which is not possible to use with ESPHome… so implementing MQTT over WebSocket in ESPHome could be a palliative to be able to use remote ESPHome devices even being in a CGNAT network (where it is not possible to expose ports) and not needing to depend on public brokers.

Marckkill, 2023-11-25:
This would be amazing.

VishveshC, 2024-05-29:
Dead thread; but did this ever come to life? Anyone find a way?

deanfourie1, 2024-05-29:
Nah, sad :(

grest, 2024-06-25:
> I manage to expose my MQTT broker through the cloudflared tunnel, but only MQTT over WebSocket, which is not possible to use with ESPHome… so implementing MQTT over WebSocket in ESPHome could be a [...]

Yes, I did the same. Support of WebSocket transport would be the excellent solution.

deanfourie1, 2024-06-27:
But seriously, how hard could it be? I don't see it requiring a lot of resources on the esp to make this feature possible.

Cloudflare have all docs available for API. Just need some smart cookie to make it happen ;)

tux789-coder, 2024-11-02:
It would be a very useful twist

x10, 2025-01-18:
This would be great

coltography, 2025-03-06:
Really wish this was possible. It would make remote devices so so so easy to set up and deploy.

mat-lo, 2025-05-29:
This would be a game changer!

deanfourie1, 2026-10-09:
Bump

Were still patiently waiting.

Or impatiently 😂

dougiteixeira, 2026-10-09:
https://github.com/orgs/esphome/discussions/3635

https://github.com/esphome/esphome/pull/16056

ankur-paan, 2026-10-10:
Thanks @dougiteixeira for mention, the community did built it and it works. due to resource constraint we were able to build only device side of it and esphome builder side is not implemented yet. I have been running it since last 3-4 months in production environment across clients and it works.

deanfourie1, 2026-10-12:
> Thanks @dougiteixeira for mention, the community did built it and it works. due to resource constraint we were able to build only device side of it and esphome builder side is not implemented yet. I have been running it since last 3-4 months in production environment across clients and it works. 

Ok thanks. Is only MQTT supported?

ankur-paan, 2026-10-12:
this enables MQTT over websocket