esphome/feature-requests #2225: Support for TM1637 6-digits displays

marek2k, 2023-05-06:
**Describe the problem you have/What new integration you would like**

Please add support for TM1637 6digits display. This modules displays [123456] in reverse order as [321654] using standard tm1637 display platform.

There is already a library that supports both 4 and 6-digits displays [https://github.com/jasonacox/TM1637TinyDisplay](https://github.com/jasonacox/TM1637TinyDisplay#tm1637-6-digit-display---tm1637tinydisplay6)

Digital-Ark, 2023-09-16:
I'm not proud of how long it took me, but I managed to get the display working like this:

```
display:
  - platform: tm1637
    clk_pin: D1
    dio_pin: D2
    lambda: |-
      // Input string here:
      std::string input1 = "123456";

      // Reorder and display input1 correctly on a 6-digit TM1637 display
      std::string reordered_input = input1.substr(2, 1) + input1.substr(1, 1) + input1.substr(0, 1)
                                  + input1.substr(5, 1) + input1.substr(4, 1) + input1.substr(3, 1);
      it.print(reordered_input.c_str());
```

Uhwah, 2024-03-14:
I'm also interested in support for this display.

satitsu, 2024-05-23:
I'm really interested in support for this kind of displays too. The TM1637 libraries in Esphome do not support 6 digits displays, as it is constated in this thread. On the other hand, going through the trick to reorder the digits that is offered here, has several issues, like displaying periods, that fails. And what's more, as the board is forced to reorder the digits on itslef, CPU and RAM consumption increases at the point that adding several variables to display (mainly sensors from home assistant for example) will eventually make the board crash and reboot. 

I'd like to ask for implementation of the right libraries.

pzacho, 2024-09-25:
I found esphome/esphome#5996 (thanks @sellerie98) introduces this feature; unfortunately is the PR still open however while waiting you could do this:
```
external_components:
  - source: github://pr#5996
    components: [ tm1637 ]
```
Remember to add the attribute `reversed: True` to the display component.

github-cyprien, 2024-12-23:
Thanks for this PR, working well for me...