esphome/feature-requests #2226: Add support for Shelly Plus Wall Dimmer

diegolrz, 2023-05-08:
**Describe the problem you have/What new integration you would like**
Add support for Shelly Plus Wall Dimmer

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

mikekornikh, 2024-03-30:
I've always been a fan of this style of wall dimmer that came with Tuya, like the Treatlife one. I just got one of these Shelly Plus wall dimmers. What interests me about it is it has an ESP32 chip in it that can do all that awesome ESPhome Bluetooth things like proxy and custom ble client commands. Also, it's only $15 at the moment on Amazon. I'd be happy to open mine up if any pics are needed for setting up a working config with this MCU.

mikekornikh, 2024-03-30:
I found the FCC page for this device, but no internal photos
https://fccid.io/2ALAYSNDM-0013US
I took some pics of it opened up, clearly seeing the ESP32 chip and the huge pads just begging to be flashed lol
[image]
[image]
[image]

mikekornikh, 2024-03-30:
Here's photos of the back of the ESP32 board and the touch board
[image]
[image]
[image]

mikekornikh, 2024-03-30:
I just found this,
Same exact board being used
https://www.elektroda.com/rtvforum/topic4014774.html

mikekornikh, 2024-03-30:
I found that someone made a script for the Gosund SW2 dimmer, which is identical except for not using an ESP32 chip like the Shelly. https://github.com/ab0tj/esphome-devices/blob/master/gosund_sw2.yaml
I'm also hoping the SW_LED_V1.2 board on the Shelly has the same layout as the SW_LED_V1.0 of the Gosund. If anyone can help me with the pin mapping, that would be amazing

pauldogg, 2024-08-01:
I wasn't able to flash anything to the device using the exposed gpio pads. Has anyone else tried?
If I can do that I'm happy to play around to get esp home or tasmota on there since Shelly seems reluctant to implement some key features. ( Dim over time, and set brightness before turning on)...

mikekornikh, 2024-08-01:
> I wasn't able to flash anything to the device using the exposed gpio pads. Has anyone else tried? If I can do that I'm happy to play around to get esp home or tasmota on there since Shelly seems reluctant to implement some key features. ( Dim over time, and set brightness before turning on)...
> 
> Edit, so sounds like the serial pins are being used....
> 
> https://www.elektroda.com/rtvforum/topic4014774.html
> 
> Gonna have to poke around some more.

When you tried to flash it, did you get it in bootloader mode first?

pauldogg, 2024-08-01:
So poking around with a continuity tester reveals that the TX and rx pins are infact connected to the TX and RX pins of the esp32.
Gpio0 doesn't seem to be connected to the gpio0 of the esp32
So that would explain why I can't get it in boot mode.
Further poking around needed

pauldogg, 2024-08-01:
It seems that the gpio0 pin can be accessed using this pad (attached pic)
[image]

 ¡UNTESTED!
I'll get around to testing this out tomorrow probably.
Stay tuned

mikekornikh, 2024-08-01:
Got my fingers crossed for you 🤞🏻 
Really awesome work

pauldogg, 2024-08-01:
Was not able to flash anything to the board....
Going to keep poking but it's not looking too good.
Any suggestions appreciated.

nagyrobi, 2024-08-01:
Another bootloader locked device? 
First Sonoff, now this?

mikekornikh, 2024-08-01:
I've never dealt with a locked ESP32 bootloader before. Would they really go through all that trouble?

nagyrobi, 2024-08-01:
EU has regulation preparations set for IoT security...

RavenSystem, 2024-08-01:
FYI: Shelly Dimmer must be powered with 5V instead 3V3 to flash it. It works in a similar way, communicating with a second MCU by UART to manage the brightness. Maybe, this device needs 5V too to be flashed.

pauldogg, 2024-08-01:
> Another bootloader locked device? First Sonoff, now this?

Welps... That would explain why I can't flash my r4 mini via pads...

pauldogg, 2024-08-01:
> FYI: Shelly Dimmer must be powered with 5V instead 3V3 to flash it. It works in a similar way, communicating with a second MCU by UART to manage the brightness. Maybe, this device needs 5V too to be flashed.

With a shot... Will give it a shot

nagyrobi, 2024-08-01:
> > Another bootloader locked device? First Sonoff, now this?
> 
> Welps... That would explain why I can't flash my r4 mini via pads...

I was able to flash the R4 minis bought 3 months ago from ali..

pauldogg, 2024-08-01:
I bought mine from Amazon a month ago. US

For the record I'm using an Adafruit flasher similar to this one https://www.adafruit.com/product/70

Could this possibly related to the 5v issue mentioned above?

nagyrobi, 2024-08-01:
I used esptool, flashed 'factory' binary.

pauldogg, 2024-08-01:
I bought mine a month ago from 

> > > Another bootloader locked device? First Sonoff, now this?
> > 
> > 
> > Welps... That would explain why I can't flash my r4 mini via pads...
> 
> I was able to flash the R4 minis bought 3 months ago from ali..

I bought mine a month ago from Amazon. US

I wonder if the 5v mentioned above is the issue. The pad says 3.3v

Also worth noting that gpio0 seems to be connected to that resistor sitting right next to it (r25?). That resistor is also connected to the 3.3v pad.

I don't know too much about electronics other than what I learn here, but if I'm not mistaken that means it's keeping it (gpio0) pulled high?

nagyrobi, 2024-08-01:
Yes that's indeed the correct purpose. You need to short gpio0 to gnd while you apply power to it, can release after. Don't put 5V to a 3.3V pin, it will damage the device. If you see a 5V pad, you can try applying it there. R4 should be flashable.

pauldogg, 2024-08-01:
> Yes that's indeed the correct purpose. You need to short gpio0 to gnd while you apply power to it, can release after. Don't put 5V to a 3.3V pin, it will damage the device. If you see a 5V pad, you can try applying it there. R4 should be flashable.

Yeah, the issue is I can't seem to pull gpio to ground.
I also tried applying ground directly to the pin on the esp chip but no success either.

The question then becomes: does one need to remove that resistor to flash it?

There's no 5v pad anywhere...

No luck flashing the r4 either...

nagyrobi, 2024-08-01:
> does one need to remove that resistor to flash it?

No.

pauldogg, 2024-08-02:
I'm all out if ideas.
Happy to try suggestions.

Thanks

pauldogg, 2024-08-24:
I have been in touch with the Shelly team via their ticket tracking website and they said "they would like to see the discussions where these features are being requested".
I pointed them to the home assistant forum, but this tells me maybe not enough people are reaching out to them.

Everyone should issue a ticket directly with Shelly so they are compelled to implement requested improvements.

vamega, 2024-09-24:
The ticketing site for Shelly is here: https://support.shelly.cloud/en/support/home

jonwest, 2024-12-22:
> I have been in touch with the Shelly team via their ticket tracking website and they said "they would like to see the discussions where these features are being requested". I pointed them to the [...]

That’s interesting. I know a bunch of people on Reddit have been asking for pretty basic functionality like being able to set the minimum brightness and they had said it was a hardware limitation, which doesn’t sound right. 

I think I’ve got a spare one of these—I’ll check and poke around and see if I can get anywhere with it via ESPHome.

pauldogg, 2024-12-24:
> because if these were able to be used similar to the MartinJerry dimmers with the physical buttons they would be awesome.

yeap.
I'm honestly shocked/frustrated they won't implement something as simple as _set brightness before turning light on_...

happy to help in any way I can.

lightingman117, 2025-01-02:
I submitted the request to Shelly too. 

But I’d prefer to flash ESPHome to them.

I’ll help however but I don’t have a ton of spare time to R&D code stuff. 

Happy to flash and test devices.