esphome/feature-requests #2227: Component for OLED Display Menu

jowgn, 2023-05-08:
It would be nice to have a Component like [lcd_menu](https://github.com/esphome/esphome/tree/dev/esphome/components/lcd_menu) for OLED displays.

otmezger, 2023-09-16:
This would be indeed really sweet. I just discovered with great disappointment that [Display Menu](https://esphome.io/components/display_menu/index.html) only works with LCD displays, leaving the pretty oled shield outside of scope.

nagyrobi, 2023-09-17:
Work is progress:
https://github.com/esphome/esphome/pull/4105

otmezger, 2023-09-17:
Amazing! Thanks.

asoltesz, 2023-10-02:
I very much second the feature request.

The OLEDs are cheap and easy to obtain so people tend to buy them instead of the more simple character LCDs.

Thus, it is important to be possible to use them in ESPHome in the simple way that Display Menu allows.

For me it wouldn't matter if there is no graphics at all (only a fixed text for each supported module) but it should work with the OLED and other pixel types.

Unfocused, 2023-12-31:
This was added in ESPHome 2023.12: 
- https://esphome.io/changelog/2023.12.0.html
- https://esphome.io/components/display_menu/graphical_display_menu

closed 2023-12-31 by ssieb