esphome/feature-requests #2235: Support for Chinese Diesel Heaters

deanfourie1, 2023-05-14:
**Describe the problem you have/What new integration you would like**
Integration for the Chinese Diesel Heater models using the RF remote control

**Please describe your use case for this integration and alternatives you've tried:**
There are no solutions, but there is already an arduino library available for this here

https://github.com/jakkik/DieselHeaterRF

**Additional context**
Would love to see this integrated in HA and ESPhome and sure there are many people who could use this integration.

jamien82, 2023-05-19:
If you haven't already check out Afterburner
http://www.mrjones.id.au/afterburner/

Bakermonitor1932, 2023-10-07:
I would also love a heater controller.

timmchugh11, 2023-10-30:
I got round to getting a (working?) custom component that reads the data from the heater, its the first custom component I have made for esphome, so could probably be much better.

https://github.com/timmchugh11/Chinese-Diesel-Heater---ESPHome

I am looking at adding control as well, but I am yet to work it out.

benek984, 2023-11-17:
100% interested in this. I would love to read and send commands to my diesel heater via RF bridge on ESPHOME

twanjaarsveld, 2023-11-26:
Did anyone get controlling the Heater using Esphome working?

djdaveb, 2023-11-27:
Im invested. I need to be able to control my heater. I'll get this installed and see what i find

deanfourie1, 2023-12-07:
> Im invested. I need to be able to control my heater. I'll get this installed and see what i find

Any update? Anxiously waiting

conor-o-brien, 2023-12-14:
I got this working tonight with tasmota using the portish firmware. The trick was adding 1 to the 5th byte to extend the time the command is sent for. I have attached the rfraw codes that worked for me. Hopefully they will work for you too. I tried with esphome but couldn't get it working
[https://drive.google.com/file/d/1i0107H01IKsyQFPWNC8vK83YkaR8_wjx/view?usp=drivesdk](url)

conor-o-brien, 2023-12-14:
This is the documentation page [https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/#portisch-firmware-specific-usage](url) for anyone that is interested.

deanfourie1, 2023-12-15:
Bugger, I don't use sonoff.

Good work!

deanfourie1, 2023-12-15:
Also, take a look at this.

https://github.com/timmchugh11/Chinese-Diesel-Heater---ESPHome

mohannaddubagh, 2023-12-15:
anyone get controlling the Heater using Esphome ?

conor-o-brien, 2023-12-16:
So I got this to work with esphome aswell using the rf bridge component [https://esphome.io/components/rf_bridge.html?highlight=bridge](url). 
This is the code I used and a screenshot of bitbucket converter.
```
esphome:
  name: rf-bridge
  friendly_name: rf-bridge

esp8266:
  board: esp01_1m

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 19200

logger:
  baud_rate: 0

rf_bridge:
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            sync: !lambda 'return format_hex(data.sync);'
            low: !lambda 'return format_hex(data.low);'
            high: !lambda 'return format_hex(data.high);'
            code: !lambda 'return format_hex(data.code);'

# Enable Home Assistant API
api:
  encryption:
    key:  !secret encryption_key
  services:
    - service: send_rf_code
      variables:
        sync: int
        low: int
        high: int
        code: int
      then:
        - rf_bridge.send_code:
            sync: !lambda 'return sync;'
            low: !lambda 'return low;'
            high: !lambda 'return high;'
            code: !lambda 'return code;'
    - service: learn
      then:
        - rf_bridge.learn

ota:
  password:  !secret ota

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
    - ssid: !secret ssid
      password: !secret wifipass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rf-Bridge Fallback Hotspot"
    password: "FJjBkF3fKvFk"

captive_portal:
    
switch:
  - platform: template
    name: "Heat On/Off"
    icon: "mdi:power"  
    id: power
    turn_on_action:
      - switch.template.publish: 
          id: power
          state: ON
      - rf_bridge.send_raw:
          raw: AAB021031801AE047E2FE42818181819090908190909081908181819081818190818181855
    turn_off_action:
      - switch.template.publish: 
          id: power
          state: OFF
      - rf_bridge.send_raw:
          raw: AAB021031801B804882FEE2818181819090908190909081908181819081818181908181855
          
button:
  - platform: template
    name: "Heat Up"
    on_press:
        - rf_bridge.send_raw:
            raw: AAB021030801B8047E2FE42818181819090908190909081908181819081818181819081855
  - platform: template
    name: "Heat Down"
    on_press:
        - rf_bridge.send_raw:
            raw: AAB021030801AE04882FEE2818181819090908190909081908181819081818181818190855
  - platform: template
    name: "Bucket sniffing"
    icon: "mdi:power"  
    id: sniffing
    on_press:
      - rf_bridge.start_bucket_sniffing
   ```
The sniffed codes need to be wrapped in json like in the screenshot

[image]

I have also included the code have sniffed in esphome here
[sniffed remote codes.txt](https://github.com/esphome/feature-requests/files/13692886/sniffed.remote.codes.txt)

pavelw, 2024-11-11:
> I got round to getting a (working?) custom component that reads the data from the heater, its the first custom component I have made for esphome, so could probably be much better.
> 
> This connects directly to the blue wire to read the data being transmitted from the controller and then the reply from the heater.
> 
> https://github.com/timmchugh11/Chinese-Diesel-Heater---ESPHome
> 
> I am looking at adding control as well, but I am yet to work it out.

Looks neat! I plan to try it out. 

The blue webasto pilot has exactly those lines needed: 
🟦 serial data
🟥 +5V
🚫 - (ground)

Ray Jones in the doc you are referring: https://gitlab.com/mrjones.id.au/bluetoothheater/-/blob/master/Documentation/V9%20-%20Hacking%20the%20Chinese%20Diesel%20Heater%20Communications%20Protocol.pdf?ref_type=heads

notes in his arduino sketch:

> - Blue wire connected to Serial 1 Rx input - preferably with series 680ohm resistor.

so I will try to connect via some resistor

mohannaddubagh, 2024-12-15:
hi i have esp01 is it work to read the data from heater
tks