esphome/feature-requests #2238: ESPHome external access

PablexXXXX, 2023-05-16:
**Describe the problem you have/What new integration you would like**
 It would be amazing to be able to especify an URL of an external home assistance instance.

**Please describe your use case for this integration and alternatives you've tried:**

This would alow the nodes to be anywhere in the world and still be able to control it from one central HA server.

**Additional context**

We already have encryption and with the Captive Portal we have everything we need to make our nodes PORTABLES.

randybb, 2023-05-16:
ESPHome acts as a server, HA is a client, so if it can connect to it (public address, or via VPN), it can be anywhere on the world.

snechiporenko, 2023-05-22:
Use VRN component [WireGuard](https://github.com/esphome/esphome/pull/4256) for do this! And [Add-on](https://github.com/hassio-addons/addon-wireguard) as server.

closed 2023-06-21 by nagyrobi