esphome/feature-requests #2240: IR Remote Climate

hiteshkher, 2023-05-16:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Hello , If wrong place Feature request I am sorry about , I am noob for GitHub
 I request for add IR Remote Climate - marq AC, ogender AC, trane AC, atomberg fan (IR Remote Controlled fan)
I am use now individual switch for AC control , Fan control , If possible please Add  Feature, If you want IR RAW data I cane give you
See attached file my dashboard for more understand, thank you !
[image]

sabya14, 2023-06-13:
@hiteshkher Can you share how you added atomberg fan support using esphome? Is it through the IR remote library? If so can you please share your config.

sabya14, 2023-07-04:
Hey @hiteshkher thanks for reaching out and helping. I would want to add that I figured a cleaner way to do this. check the following config out.

`
 output:
  - platform: template
    id: fan_speed_output
    type: float
    write_action:
      - logger.log:
          format: "Speed to set: %i"
          args: [ 'id(fan_atomberg).speed']
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 1);
          then:
            - logger.log: "Setting speed to 1"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote
                address: 0xF300
                command: 0x748B
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 2);
          then:
            - logger.log: "Setting speed to 2"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x6F90
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 3);
          then:
            - logger.log: "Setting speed to 3"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x758A
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 4);
          then:
            - logger.log: "Setting speed to 4"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x6C93
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 5);
          then:
            - logger.log: "Setting speed to 5"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x7788
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 6);
          then:
            - logger.log: "Setting speed to 6"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x708F

# Individual switches
fan:
  - platform: speed
    output: fan_speed_output
    name: "Atomberg Fan"
    id: fan_atomberg
    speed_count: 6
    restore_mode: ALWAYS_OFF
    on_turn_on:
      - logger.log: "Fan Turned On!"
    on_turn_off:
      - logger.log: "Fan Turned Off!"
      - remote_transmitter.transmit_nec:
          transmitter_id: remote
          address: 0xF300
          command: 0x6E91
 `
 
 The advantage here is that it will work as a speed control fan

hiteshkher, 2023-07-04:
Thank you for improvement code, I will sure try

hiteshkher, 2023-07-04:
I have not much knowledge about basic code and noob for lamda so i was use simple code

hiteshkher, 2023-07-05:
captive_portal:
  
# IR #

remote_transmitter:
  pin:
    number: D1
  carrier_duty_percent: 50%

# TEMPERATURE #

sensor:
  - platform: dht
    pin: D3
    temperature:
      name: "dht11 Temperature"
    humidity:
      name: "dht11 Humidity"
    update_interval: 60s

# MANUAL SWITCH #

binary_sensor: 

  - platform: gpio
    name: fan
    pin:
      number: D2
      inverted: true
      mode:
        input: true
        pullup: true

    on_press:
      then:
      - switch.turn_on: fan1
    on_release:
      then:  
      - switch.turn_off: fan1
#-------------------------------------------------2 end input 
  - platform: gpio
    name: light
    pin:
      number: D5
      inverted: true
      mode:
        input: true
        pullup: true

    on_press:
      then:
      - switch.turn_on: light
    on_release:
      then:  
      - switch.turn_off: light

#-------------------------------------------------3 end input 

  - platform: gpio
    name: night lamp
    pin:
      number: D6
      inverted: true
      mode:
        input: true
        pullup: true
    
    on_press:
      then:
      - switch.turn_on: nightlamp
    on_release:
      then:
      - switch.turn_off: nightlamp

#-------------------------------------------------4 end input 

# RELAY OUTPUT #

switch:
  - platform: gpio  
    name: fan
    id: fan1
    pin: D4        
    #inverted: true
  #---------------------------------------------    
  - platform: gpio  
    name: light
    id: light
    pin: D7       
    #inverted: true
  #---------------------------------------------   
  - platform: gpio  
    name: nightlamp
    id: nightlamp
    pin: D0
    #inverted: true

# IR CODE #

  - platform: template
    name: "1"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x748B
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "2"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x6F90
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "3"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x758A
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "4"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x6C93
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "5"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x7788
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "BOOST"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x708F
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "fan_light"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x6897
          repeat:
            times: 1
            wait_time: 50ms     

output:
  - platform: template
    id: fan_speed_output
    type: float
    write_action:
      - logger.log:
          format: "Speed to set: %i"
          args: [ 'id(fan_atomberg).speed']
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 1);
          then:
            - logger.log: "Setting speed to 1"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote
                address: 0xF300
                command: 0x748B
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 2);
          then:
            - logger.log: "Setting speed to 2"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x6F90
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 3);
          then:
            - logger.log: "Setting speed to 3"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x758A
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 4);
          then:
            - logger.log: "Setting speed to 4"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x6C93
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 5);
          then:
            - logger.log: "Setting speed to 5"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x7788
      - if:
          condition:
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 6);
          then:
            - logger.log: "Setting speed to 6"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x708F

# Individual switches
fan:
  - platform: speed
    output: fan_speed_output
    name: "Atomberg Fan"
    id: fan_atomberg
    speed_count: 6
    restore_mode: ALWAYS_OFF
    on_turn_on:
      - logger.log: "Fan Turned On!"
    on_turn_off:
      - logger.log: "Fan Turned Off!"
      - remote_transmitter.transmit_nec:
          transmitter_id: remote
          address: 0xF300
          command: 0x6E91

hiteshkher, 2023-07-05:
can you arrange my code and back ?  I got some error

hiteshkher, 2023-07-19:
> Hey @hiteshkher thanks for reaching out and helping. I would want to add that I figured a cleaner way to do this. check the following config out.
>
> ```
>  output:
>   - platform: template
>     id: [...]

@sabya14 thank you for code , working parfect

ahsh94, 2023-09-18:
Hii guys, 

I am sorry i don’t know anything about making codes, but i know how to download using codes and tweak ios apps lol. 

My dumb question is the next, will this work is i run it through an application using Iphone only ? To make some sort of a remote to my General AC, the original one went down as a victim in a middle of a Fifa Match game.

Thanks already.

DCG81, 2024-05-13:
hey @hiteshkher I have IR remote running on ESP Home where I have setup individual IR codes for each button press for mt Atomberg fan

below is my yaml code, how do change it as a Speed Control Fan like what you have mentioned above, pls help

```

esphome:
  name: bedroom-remoteir
  friendly_name: IR Remote
  
esp8266:
  board: esp01_1m

captive_portal:

# Enable logging
logger:

api:
  
ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 50%

remote_receiver:
  id: rcvr
  pin: 
    number: GPIO05
    inverted: true
    mode: INPUT_PULLUP
  tolerance: 50%
  dump: raw

sensor:
 - platform: wifi_signal
   name: "IR Signal Strength"
   update_interval: 60s
 
 - platform: uptime
   name:  "IR Remote Uptime"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "IR IP Address"

switch:
  - platform: restart
    name: "IR Remote Reboot"
    
  - platform: template
    name: Fan Power
    id: fan_power_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [ 9051, -4465, 611, -529, 610, -528, 610, -530, 610, -529, 610, -530, 610, -530, 607, -531, 611, -529, 610, -1651, 609, -1652, 612, -527, 611, -529, 611, -1650, 611, -1649, 611, -1651, 609, -1653, 580, -1680, 611, -529, 609, -530, 610, 
-529, 610, -1651, 612, -529, 608, -532, 608, -1650, 611, -528, 611, -1651, 611, -1650, 613, -1648, 610, -529, 610, -1651, 610, -1651, 611, -529, 611]

  - platform: template
    name: Fan Speed 1
    id: fan_1_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9052, -4481, 595, -545, 597, -544, 593, -544, 594, -546, 620, -519, 592, -547, 594, -545, 596, -544, 594, -1667, 622, -1639, 647, -493, 593, -546, 593, -1642, 622, -1662, 594, -1669, 594, -1642, 620, -1666, 598, -1662, 596, -546, 594, 
-1666, 594, -545, 594, -545, 621, -519, 621, -1639, 596, -543, 599, -540, 620, -1642, 594, -546, 619, -1641, 620, -1640, 621, -1640, 596, -545, 596]

  - platform: template
    name: Fan Speed 2
    id: fan_2_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9029, -4457, 622, -543, 592, -547, 621, -519, 593, -546, 592, -547, 620, -520, 593, -545, 596, -544, 620, -1640, 593, -1668, 648, -492, 595, -545, 594, -1666, 648, -1588, 615, -1647, 619, -1665, 596, -546, 620, -519, 596, -543, 620, 
-521, 592, -1667, 595, -545, 620, -519, 570, -1692, 618, -1643, 593, -1667, 597, -1665, 592, -1668, 596, -544, 594, -1667, 620, -1642, 592, -549, 594]

  - platform: template
    name: Fan Speed 3
    id: fan_3_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9045, -4471, 607, -531, 608, -532, 581, -557, 611, -530, 582, -556, 610, -530, 583, -556, 611, -529, 606, -1656, 611, -1649, 582, -558, 582, -558, 607, -1655, 607, -1656, 579, -1679, 608, -1654, 608, -533, 608, -1652, 584, -556, 610, 
-1651, 608, -531, 610, -531, 581, -559, 607, -1652, 609, -1652, 610, -532, 580, -1680, 582, -556, 582, -1680, 609, -1654, 606, -1653, 609, -531, 608]

  - platform: template
    name: Fan Speed 4
    id: fan_4_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9025, -4493, 612, -528, 607, -533, 609, -531, 581, -557, 608, -533, 609, -552, 561, -557, 608, -531, 608, -1653, 581, -1681, 607, -532, 610, -530, 607, -1653, 583, -1679, 611, -1652, 605, -1655, 609, -1654, 609, -1652, 583, -556, 610, 
-529, 610, -1651, 608, -533, 606, -532, 610, -1652, 583, -557, 610, -527, 613, -1650, 610, -1650, 610, -531, 610, -1650, 610, -1651, 611, -530, 582]

  - platform: template
    name: Fan Speed 5
    id: fan_5_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9060, -4457, 621, -545, 596, -542, 599, -542, 621, -518, 646, -495, 596, -543, 618, -522, 620, -519, 597, -1641, 620, -1644, 640, -523, 617, -522, 594, -1667, 646, -1590, 623, -1666, 595, -1640, 645, -522, 623, -516, 621, -519, 621, 
-1617, 620, -545, 618, -494, 623, -543, 622, -1640, 622, -1641, 595, -1666, 598, -1663, 624, -517, 594, -1666, 619, -1619, 620, -1666, 622, -519, 621]

  - platform: template
    name: Fan Speed Boost
    id: fan_boost_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9057, -4459, 646, -520, 594, -545, 592, -548, 594, -546, 594, -546, 592, -548, 620, -519, 645, -496, 595, -1666, 620, -1617, 620, -544, 624, -517, 619, -1620, 641, -1620, 619, -1667, 596, -1667, 595, -1667, 623, -1641, 595, -1666, 596, 
-1666, 621, -520, 619, -520, 599, -541, 646, -1617, 619, -521, 594, -545, 595, -546, 594, -545, 593, -1669, 596, -1641, 617, -1671, 595, -545, 620]

  - platform: template
    name: Fan Light
    id: fan_light_button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9021, -4493, 610, -531, 607, -531, 610, -530, 582, -557, 610, -556, 583, -534, 606, -531, 608, -530, 609, -1652, 584, -1678, 583, -557, 609, -531, 609, -1650, 611, -1651, 610, -1653, 608, -1650, 583, -1682, 605, -1655, 608, -1653, 608,
-531, 610, -1652, 609, -529, 609, -530, 608, -1654, 607, -532, 608, -533, 607, -531, 610, -1650, 605, -535, 608, -1652, 608, -1653, 609, -531, 610]    

```

hiteshkher, 2024-05-13:
> hey @sabya14 @hiteshkher I have IR remote running on ESP Home where I have setup individual IR codes for each button press for my Atomberg fan
> 
> below is my yaml code, how do change it as a Speed Control Fan like what you have mentioned above, pls help
> 
> ```
> 
> esphome:
>   name: bedroom-remoteir
>   friendly_name: IR Remote
>   
> esp8266:
>   board: esp01_1m
> 
> captive_portal:
> 
> # Enable logging
> logger:
> 
> api:
>   
> ota:
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> remote_transmitter:
>   pin: GPIO14
>   carrier_duty_percent: 50%
> 
> remote_receiver:
>   id: rcvr
>   pin: 
>     number: GPIO05
>     inverted: true
>     mode: INPUT_PULLUP
>   tolerance: 50%
>   dump: raw
> 
> sensor:
>  - platform: wifi_signal
>    name: "IR Signal Strength"
>    update_interval: 60s
>  
>  - platform: uptime
>    name:  "IR Remote Uptime"
> 
> text_sensor:
>   - platform: wifi_info
>     ip_address:
>       name: "IR IP Address"
> 
> switch:
>   - platform: restart
>     name: "IR Remote Reboot"
>     
>   - platform: template
>     name: Fan Power
>     id: fan_power_button
>     turn_on_action:
>       - remote_transmitter.transmit_raw:
>           carrier_frequency: 38kHz
>           code: [ 9051, -4465, 611, -529, 610, -528, 610, -530, 610, -529, 610, -530, 610, -530, 607, -531, 611, -529, 610, -1651, 609, -1652, 612, -527, 611, -529, 611, -1650, 611, -1649, 611, -1651, 609, -1653, 580, -1680, 611, -529, 609, -530, 610, 
> -529, 610, -1651, 612, -529, 608, -532, 608, -1650, 611, -528, 611, -1651, 611, -1650, 613, -1648, 610, -529, 610, -1651, 610, -1651, 611, -529, 611]
> 
> 
>   - platform: template
>     name: Fan Speed 1
>     id: fan_1_button
>     turn_on_action:
[... 134 lines cut ...]
            lambda: return (id(fan_atomberg).state == 1 and id(fan_atomberg).speed == 6);
          then:
            - logger.log: "Setting speed to 6"
            - remote_transmitter.transmit_nec:
                transmitter_id: remote

                address: 0xF300
                command: 0x708F

# Individual switches
fan:
  - platform: speed
    output: fan_speed_output
    name: "Atomberg Fan"
    id: fan_atomberg
    speed_count: 6
    restore_mode: ALWAYS_OFF
    on_turn_on:
      - logger.log: "Fan Turned On!"
    on_turn_off:
      - logger.log: "Fan Turned Off!"
      - remote_transmitter.transmit_nec:
          transmitter_id: remote
          address: 0xF300
          command: 0x6E91

DCG81, 2024-05-13:
hey @hiteshkher thanks a ton, it is working perfectly now as a Fan with Speed controls... I was little confused with the codes, all I had to do was to rename the transmitter_id correctly

I have couple of questions

1) restore_mode: ALWAYS_OFF - does thie mean when there is power failure fan will go to off mode after power restore ? 

2). Atomberg fan has a LED light, can that light switch be enabled thru code?

hiteshkher, 2024-05-13:
1. power failure and back arrive then fan should last status follow ( now I dont remember about , i am not sure working or not )
2. YES

hiteshkher, 2024-05-13:
@DCG81  If you want individual switches , use this ymal

```
switch:
  - platform: gpio  
    name: fan
    id: fan1
    pin: D4        
    #inverted: true
  #---------------------------------------------    
  - platform: gpio  
    name: light
    id: light
    pin: D7       
    #inverted: true
  #---------------------------------------------   
  - platform: gpio  
    name: nightlamp
    id: nightlamp
    pin: D0
    #inverted: true

# IR CODE #

  - platform: template
    name: "1"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x748B
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "2"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x6F90
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "3"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x758A
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "4"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x6C93
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "5"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x7788
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "BOOST"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x708F
          repeat:
            times: 10
            wait_time: 50ms
  - platform: template
    name: "fan_light"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xF300
          command: 0x6897
          repeat:
            times: 1
            wait_time: 50ms     
```

DCG81, 2024-05-13:
Thanks @hiteshkher I got the Fan Light enabled.

I was curious did u get the IR Remote Climate working with the raw codes ?

hiteshkher, 2024-05-13:
> Thanks @hiteshkher I got the Fan Light enabled.
> 
> I was curious did u get the IR Remote Climate working with the raw codes ?

Nope

radiantly, 2024-06-16:
Thanks @sabya14 @hiteshkher - I've got my atomberg fan working with the IR code posted above.

Here's my config: https://gist.github.com/radiantly/54cea0c55bc4265e10130db4f0e48d15