esphome/feature-requests #2241: Problem with interfacing 3x PZEM004T V1.0 modules to a WT32-ETH01 board.

tux-at-chello, 2023-05-18:
In the past we used an Arduino Micro with an ethernet shield an 3 comports
to interface 3 PZEM004T V1.0 modules and an Arduino based self written program
to transfer data from a 3 phase solar panel array into a crestron controller to
generate some data and graphs about the production of this 40 panel solar array.

Now in the days of ESPHome and Home Assistand we decided to convert these measuring
modules over to an WT32-ETH01 module, just for the reason of multiple serial ports 
and intergrated ethernet, making it a hard wired interface and avoiding WiFi traffic
at a high load due to the frequent repeat of the measurements.

For this purpose I defined 3 seial ports on the WT32-ETH01, using the 'receive only'
ports as the RX ports of the construction, leading up to the following layout.

Phase L1 : TX data on GPIO 17, RX data on GPIO05
Phase L2 : TX data on GPIO 13, RX data on GPIO35
Phase L3 : TX data on GPIO 15, RX data on GPIO36

Standard TX0 and RX0 are left free, to be able to program the unit via serial protocol
in case the ethernet interface fails and 'OTE' ( over the Ethernet ) is not available.

For testing purposes I have 2 push buttons to ground on GPIO2 and GPIO4 to have some
inputs to play around with  ( and testing connectivity to ESPhome ) in further projects.

This gives me the following YAML file.

- - - - - - - - - - 8< - - - - YAML snip starts here - - - - 8< - - - - - - - -	

substitutions:
  devicename: eth01
  long_devicename: eth01

esphome:
  name: eth01

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

web_server:
  port: 80

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO2
      mode: INPUT_PULLUP
      inverted: True
    name: "$long_devicename button_1"
    on_press:
      - logger.log: "$long_devicename button_1"
  - platform: gpio
[... 369 lines cut ...]
[21:58:07][C][api:143]:   Using noise encryption: NO
[21:58:07][D][sensor:110]: 'PZEM-004T Energy L1': Sending state 7486489.00000 Wh with 0 decimals of accuracy
[21:58:07][D][pzem004t:078]: Got Energy 7486489 Wh
[21:58:11][D][sensor:110]: 'PZEM-004T Voltage L1': Sending state 225.00696 V with 1 decimals of accuracy
[21:58:11][D][pzem004t:052]: Got Voltage 234.7 V
[21:58:12][D][sensor:110]: 'PZEM-004T Current L1': Sending state 0.00000 A with 2 decimals of accuracy
[21:58:12][D][pzem004t:061]: Got Current 0.00 A
[21:58:12][D][sensor:110]: 'PZEM-004T Power L1': Sending state 0.00000 W with 0 decimals of accuracy
[21:58:12][D][pzem004t:069]: Got Power 0 W
[21:58:13][D][sensor:110]: 'PZEM-004T Energy L1': Sending state 7486489.00000 Wh with 0 decimals of accuracy
[21:58:13][D][pzem004t:078]: Got Energy 7486489 Wh
[21:58:21][D][sensor:110]: 'PZEM-004T Voltage L1': Sending state 224.62300 V with 1 decimals of accuracy
[21:58:21][D][pzem004t:052]: Got Voltage 234.3 V
[21:58:22][D][sensor:110]: 'PZEM-004T Current L1': Sending state 0.00000 A with 2 decimals of accuracy
[21:58:22][D][pzem004t:061]: Got Current 0.00 A
[21:58:22][D][sensor:110]: 'PZEM-004T Power L1': Sending state 0.00000 W with 0 decimals of accuracy
[21:58:22][D][pzem004t:069]: Got Power 0 W
[21:58:23][D][sensor:110]: 'PZEM-004T Energy L1': Sending state 7486489.00000 Wh with 0 decimals of accuracy
[21:58:23][D][pzem004t:078]: Got Energy 7486489 Wh

- - - - - - - - - - 8< - - - - working bootstrap snip ends here - - - - 8< - - - - - - - -	

Question : are we doing anything illegal to the WTH32-ETH01 or is anything else causing this problem ?

Kind regards from an enthousiastic group of 4 out of The Netherlands.

ssieb, 2023-05-18:
An ESP32 has 3 hardware uarts and esphome doesn't support software serial on it.
Previously, you could only use uart0 if you use the default pins (1 and 3), but that was fixed in the latest release, so if you upgrade esphome, it should work now.

tux-at-chello, 2023-05-19:
Hello,

I did some testing today and found an error message on the serial debug port with the esphome  version 2023.4.4 : 

```
[15:44:26][I][app:029]: Running through setup()...
[15:44:26][C][uart.arduino_esp32:077]: Setting up UART...
[15:44:26][C][uart.arduino_esp32:077]: Setting up UART...
[15:44:26][C][uart.arduino_esp32:077]: Setting up UART...
[15:44:26][    68][E][HardwareSerial.cpp:309] begin(): Serial number is invalid, please use numers from 0 to 2
```

After upgrading to version 2023.5.1 and recompiling the code the message changed into : 

```
[I][app:029]: Running through setup()...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][uart.arduino_esp32:077]: Setting up UART...
[C][uart.arduino_esp32:077]: Setting up UART...
[W][uart.arduino_esp32:104]: Maximum number of UART components created already.
[E][component:113]: Component uart was marked as failed.
```

Even after I changed the tx and rx pins to GPIO1 and GPIO3 ( the debug port ) for the third UARD definition.
So software-serial is ( still ) not implemented in esphome version 2023.5.1 and we are limited to two ports?

tux-at-chello, 2023-05-19:
Sorry for the mess: this issue is still open !

ssieb, 2023-05-19:
You have to disable the serial logging.  There is no plan to add software serial to ESP32, there are 3 uarts available.