esphome/feature-requests #2242: ESP Wifi Programmer

jamien82, 2023-05-19:
**Describe the problem you have/What new integration you would like**

The ability to have a battery powered esp device that can programme another device via OTA

**Please describe your use case for this integration and alternatives you've tried:**

The use case is if a device is mounted up a pole and cant be reached by OTA then you could reprogram it easily

**Additional context**

nagyrobi, 2023-05-19:
OTA ESP to ESP?

Mat931, 2023-06-25:
A couple years ago I started working on something similar for flashing Arduinos (ATmega328p) wirelessly using ESPHome, based on [ESP_AVRISP](https://github.com/lbernstone/ESP_AVRISP). Flashing other ESPs in this way could be useful too!

rob040, 2023-08-23:
@jamien82 How do you imagine how to select the **right** image to OTA program into the **right** target, without an UI and without Wifi connection to HA or PC?  Where would that image be stored?

Why would an ESPhome device NOT be reachable via Wifi and still be functional in an HA / IOT environment?

The described use case feels to me a bit weak.
On the other hand, #2283 (closed) FR, where a (battery powered) ESPhome device prepares a NON-ESPhome device for ESPhome OTA usage by (wired-) programming a bare image using an esp-tool alike upload protocol. Similar to @Mat931 mentions above done for AVR.
However, using a PC for this, the current regular way, may still be handier for its speed, direct feedback and readable logging output on its display, which the here requested headless solution here requested doesn't have.

(in the past, I had some MCU programmers that also had the capability to store a programming image inside, and program a target at a button press. I must say I and my coworkers never used that feature).