esphome/feature-requests #2250: ESPiLight Support - 433 MHz devices. reopened #315

Micha10, 2023-05-24:
**Describe the problem you have/What new integration you would like**

Please integrate espilight to esphome. 
You wrote in the other issue, that you want a sepcific protcol-wish, but imho that's not the right way. So I wish quigg_gt1000, 12 days later another person wishes quigg_gt7000, days later the next wish. I'm sure you don't want to implement everything one by one if you can have all at the same time.

Espilight worked hard to extract the protocols from pilight for use in esp8266 and esp32. That seems perfect fitting to esphome.

**Please describe your use case for this integration and alternatives you've tried:**
I spent 2 weeks to try to get a brennenstuhl switch get working with unit/id and raw data. The weired protocol seems not supported by any library than pilight. 

Espilight's wrapper have just two parameters. The first is the protocol and the second is the json representation of the configuration, i.E.
`rf.send("quigg_gt9000", "{\"id\":131271,\"unit\":1,\"state\":\"on\"}");`

I suggest not to try to integrate it in the existing transmitter-protocol, but to add the feature as new one. 
I'm really fine if I could use the esphome transmitter OR espilight transmitter... not both the same time.

**Additional context**
Perhaps you can add just a new device "espilight" with 2 parameters. Inside you can add the espilight as submodule.

If you have any futher questions about that issue don't hesitate to contact me.

Blackspell01, 2023-06-29:
any news yet?

Micha10, 2023-06-30:
I've added espilight as external component. That's working for me, but an integration would be cool, nevertheless

You can crib from:
https://github.com/Micha10/esphome_espilight
https://github.com/Micha10/espilightsw

Blackspell01, 2023-07-02:
honestly Im not really an expert at this topic, can you please explain what I have to do? When I try to compile the yaml file it throws errors with both links you provided (i put the .h files in the esphome directory respectivly). Will this work on an esp8266 based device or only for esp32 and how do i specify my gpio pins for the receiver and transmitter? My goal is to sniff a quigg_gt9000 code in order to replicate it. Thank you

Micha10, 2023-07-28:
Could you show me the error you got. I'll try to help, but I'm also no expert. It took weeks to get that working.

Montreal666, 2023-10-04:
Any hope on this one?
Also interested to see Pilight supported; Currently using OpenMQTTGateway + Sonoff RF Bridge + Pilight, however auto discovery is not supported so would definitely migrate to ESPHOME if support is added.
Watching. 
Thanks!