esphome/feature-requests #2251: Please add support for the Hailingke 2450 radar module.

00660, 2023-05-25:
[HLK LD2450 1T2R运动目标检测追踪模组说明书 V1.00.pdf](https://github.com/esphome/feature-requests/files/11560732/HLK.LD2450.1T2R.V1.00.pdf)

lboue, 2023-05-25:
[HLK-LD2450 : 24Ghz one-transmitter-two-receiver human motion tracking trajectory radar module Wall mounted installation
](https://hlktech.net/index.php?id=1157)

### Product Overview
Motion target tracking is to track the position of the moving target in the region in real time, and realize the distance, angle and speed measurement of the moving target in the region.The LD2450 is a motion target tracking sensor module from the Hi-link 24G millimetre wave radar series, which includes extremely simplified 24 GHz radar sensor hardware and intelligent algorithm firmware. The solution is mainly used in general indoor scenarios such as homes, offices and hotels to enable the location tracking of moving human bodies. 
The sensor hardware consists of an AloT millimetre wave radar chip, a high performance one-transmitter-two-receiver microstrip antenna and a low cost MCU and peripheral auxiliary circuitry. The intelligent algorithm firmware uses FMCW waveforms and the radar chip's proprietary advanced signal processing technology. 
It supports serial output of detection data, which is plug-and-play and can be flexibly applied to different smart scenarios and end products.

[image]

dwildstr, 2023-10-15:
There are two extant projects which seem to provide a promising starting point: [the Screek human sensor 2A configuration](https://github.com/screekworkshop/screek-human-sensor/blob/main/2a/yaml/human-sensor-2a-stable-github.yaml), which is a YAML file with a lot of code in lambdas, and [tsunglung's ld2450 custom component](https://github.com/tsunglung/esphome-ld2450), which relegates a lot of the serial-port processing to a custom component but which still relies on a medium-messy lambda in the YAML. Neither are exactly a drop-in `ld2450:` component which painlessly provides a bunch of sensors but both have the back end talking-to-an-LD2450 aspect figured out.

bzumik1, 2023-10-18:
Hi, how does it look, is there any plan to add it to ESPHome codebase?

B48D81EFCC, 2023-10-21:
Isn't it the same as: #2219 ?

Want100Cookies, 2023-11-01:
I recently purchased the [Everything Presence Lite](https://shop.everythingsmart.io/en-nl/products/everything-presence-lite) which features this sensor. The [default configuration](https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/everything-presence-lite-ha.yaml) comes with a large lambda parsing lines from serial. I just found out that [amcfague](https://github.com/amcfague) is working on proper support in [esphome/esphome#5624](https://github.com/esphome/esphome/pull/5624). Thank you Andrew, keep up the good work!

bzumik1, 2023-11-01:
Thanks for update, that sounds great!

alexruffell, 2024-06-06:
It looks like official support will be added with this pull, but it is taking months :(

https://github.com/esphome/esphome/pull/5674

Does this FR need to stay open?