esphome/feature-requests #2254: Add support for Inkplate 2

Zeik0s, 2023-05-27:
**Describe the problem you have/What new integration you would like**

Please add support for the Inkplate 2 Display. 

**Please describe your use case for this integration and alternatives you've tried:**

Using this board with native esphome would be much easier and could offer more options as if it's only using the arduino library. I already tried setting it up with esphome, but there is no matching config for controlling the display, as it is some different e-ink display. Also the existing waveshare models aren't working with this display. 

**Additional context**

Inkplate 2 arduino library => [pin mapping](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/master/src/boards/Inkplate2.h)
[Hardware Design Docs](https://github.com/SolderedElectronics/Soldered-Inkplate-2-hardware-design)

Zeik0s, 2023-05-31:
@brumouta please add a thumbs up at the issue and **not**  post +1!

paviro, 2024-02-10:
@Zeik0s do you mean the Inkplate 10 v2? From my perspective this works already at least I am using it with ESPHome.

sylwester-, 2025-01-31:
Has anyone managed to do this?

Zeik0s, 2025-01-31:
I haven't seen anything yet. I thought i might try it myself, but have never built something around esphome and didn't have time to do it.