esphome/feature-requests #2257: Sprinkler Controller to Support Both a Main Valve and a Pump

rjmcreavy, 2023-06-01:
**Describe the problem you have/What new integration you would like**
In the Sprinkler Controller, I would like to specify both a main valve and a pump. Currently can specify one pump_switch_id (or pump_on_switch_id/pump_off_switch_id), which controls a pump or a main valve. I would like to specify both a pump and a main valve.

**Please describe your use case for this integration and alternatives you've tried:**
My irrigation system is feed from a rain barrel, so it has a main valve and a pump (because of low pressure from the barrel).  Downstream are zone valves.  Both the main value and the pump must be turned on.  The main valve is a latching valve, the pump is simple on/off.  Main valve and pump are controlled through different GPIO.  Using the Sprinkler Controller, I see no direct way to turn on both the latching main valve, and the pump, when turning on the zones.

I've though about using a template switch for the pump_switch_id (not sure if it will work), but would prefer it natively in the Sprinkler Controller.

**Additional context**

nagyrobi, 2023-06-03:
@kbx81

kbx81, 2023-06-03:
> I've though about using a template switch for the pump_switch_id (not sure if it will work), but would prefer it natively in the Sprinkler Controller.

You don't need a template switch for this, just an additional GPIO pin/switch for your pump (as you suggested you already have).

Considering the example [here](https://esphome.io/components/sprinkler.html#single-controller-three-latching-valves-single-latching-pump), all you need to do is add an `on_turn_on` ([docs](https://esphome.io/components/switch/index.html#switch-on-turn-on-off-trigger)) trigger to each of the GPIO switches you already have for your main (latching) valve:

```yaml
switch:
  - platform: gpio
    id: sprinkler_pump
    pin: 13
  - platform: gpio
    id: sprinkler_main_valve_off
    pin: 14
    on_turn_on:
      switch.turn_off: sprinkler_pump
  - platform: gpio
    id: sprinkler_main_valve_on
    pin: 15
    on_turn_on:
      switch.turn_on: sprinkler_pump
```

dala318, 2023-06-05:
The on_turn_on/off actions in comment above work when you always need both valve and pump on/off in sync.
In my case the main valve enables water to all zone-valves but I also have a pump that can boost water pressure to whole system. But this boost is only needed for some zones.

I would prefer the integration keeping track of when which of these switches need to be on by providing a list of pump_switch_id per zone, etcher just the main_valve or both main_valve & main_pump.

This can of course be solved with some template switches. But it's also a bit of pain to work out the timing between them (dependant of the overlap settings), specially since I don't ever want the pump to run when the main_valve is off, risking it running dry and overheating.

Safest option I currently see is having just two "empty, optimistic" template switches which on every on_turn_on/on_turn_off action run a script that sets the GPIO switches based on the state of the templates.

kbx81, 2023-06-06:
Ah, this is an interesting use case. I'm curious, is there a commercial controller that accommodates this? Idle curiosity because I've never heard of a scenario where (I'll paraphrase) sometimes I want the pump to run but other times I don't. 😇

If there's a consistent relationship between which zone(s) need the pump and which zones do not, why not just use two sprinkler controller instances (where one activates the pump and other does not)? You could add an action to the main switch that switches off the other controller in the event someone attempts to activate both simultaneously.

I've got quite a few things on my plate at the moment so I probably won't get back around to working on the sprinkler controller for a while. Adding this functionality won't be terribly complex but it also won't be trivial, either.

nagyrobi, 2023-06-06:
Or an `if` could be added to @kbx81's example to turn pump on together with the main valve ony if switches corresponding to those certain zones is on.

rjmcreavy, 2023-06-07:
Thanks @kbx81this is very helpful.

dala318, 2023-06-08:
Hi, @kbx81, not sure how common this setup is and solvable be some templating on the switches as I and other have proposed. Not urgent for my sake but more of a "nice to have".

@nagyrobi, have not tested your solution but have to ensure the on/off commands to the pump_switch is called every zone-progress, even if already in the expected state.

cdk222, 2023-10-07:
The above use of 'on_turn_on' and linking it to the pump GPIO does cause an issue if you are also using the pump delay feature. 
If you have a delay enabled then the Main Valve will only open when the Pump delay is finished and subsequently 'Turns On' which defeats the purpose of the delay.

Is there a way that we could add optional config of a "main_valve_switch_id" into the valve section of sprinkler config? 
The reason for this is that it would need behave and obtain the complete sprinkler cycle duration in the same way as pump_switch_id .......however it would need to not be impacted by the pump_**_delay

***Note I did try a few other methods of linking the Main valve to individual valve start's but each time I failed due to not being able to effectively get the total cycle time for all valves running in the session. Especially when introducing delays and valve overlaps***

roykrikke, 2025-06-29:
@kbx81 I have the same wish as @rjmcreavy. I also encountered the same challenge as @cdk222

Any tip is appreciated to get this working properly. Example(s) would be absolutely awesome.

**My goal:**
I would like to have the Sprinkler Controller to somehow open my Main Valve and wait for e.g. 3 seconds and the open a zone value wait for e.g. 3 seconds and then enable my Pump.

In the Sprinkler Controller, I would like to specify both a main valve and a pump. Currently can specify one pump_switch_id, which controls a pump or a main valve. I would like to specify both a pump and a main valve.

My irrigation system is feed from main watersource, so it has a main valve and a pump (because of low pressure from the watersource). Downstream are zone valves. The main valve, a zone valve and the pump must be turned on to get my system to work well.  Main valve and pump are controlled through different GPIO (because of delay requirement). Using the Sprinkler Controller, I see no direct way to turn on both the main valve, and the pump, when turning on the zones. 

This requires the following steps:
Starting:
1: open main valve
2: open zone valve (with a delay)
3: enable pump (with a delay)

Stopping:
1: Disable pump
2: close zone valve (with a delay)
3: close main valve (with a delay)

I've though about using a template switch for the pump_switch_id (not sure if it will work), but would prefer it natively in the Sprinkler Controller.

Snippets from my currect config:

```
#############################
#  Establish Substitutions  #
#############################
substitutions:
  devicename: irrigation_controller
  system_name: irrigation-controller
  friendly_name: "Irrigation Controller"
  device_description: "The T-Relay is a handy relay module with Wi-Fi and Bluetooth. This module contains 4 relays and an ESP32 Wrover"
  zone1: "Zone1"
  zone2: "Zone2"
  zone3: "Zone3"
  zone4: "Zone4"
  zone1_id: valve_0
  zone2_id: valve_1
  zone3_id: valve_2
  zone4_id: valve_3
  sensor_update_frequency: 1s
  
switch:
  - platform: gpio
    pin:
      number: GPIO33
      inverted: false
    internal: true
    name: "${friendly_name} Pump"
    id: ${devicename}_pump
	
sprinkler:
  - id: $devicename
    main_switch: "Start/Stop"
    auto_advance_switch: "Auto Advance"
    reverse_switch: "Reverse"
    next_prev_ignore_disabled: true

    multiplier_number:
      id: run_time_multiplier
      name: "Run Time Multiplier"
      min_value: 0.1
      max_value: 5
      step: 0.1
      icon: mdi:multiplication
      mode: BOX
      restore_value: true
      initial_value: 0.1

    # Causes pump start/stop delay for the duration specified.
    # Causes the current valve and the next valve to run simultaneously for the duration specified.
    # Prevents water hammer effect.
    pump_start_pump_delay: 3s
    pump_stop_valve_delay: 3s
    valve_overlap: 3s

    valves:
################
#  Zone One    #
################
    - valve_switch:
        id: $zone1_id
        name: $zone1
        on_turn_on:
        - then:
          - text_sensor.template.publish:
              id: irrigation_controller_status
              state: $zone1 Active
        on_turn_off:
        - then:
          - text_sensor.template.publish:
              id: irrigation_controller_status
              state: Idle
      enable_switch: Enable ${zone1}
      run_duration_number:
        id: zone_1_duration
        name: ${zone1} Duration
        min_value: 1
        max_value: 60
        step: 1
        unit_of_measurement: min
        icon: mdi:timer-outline
        mode: BOX
        restore_value: true
        initial_value: 1
      pump_switch_id: ${devicename}_pump
      valve_switch_id: relay1

################
#  Zone Two    #
################
    - valve_switch:
        id: $zone2_id
        name: $zone2
        on_turn_on:
        - then:
          - text_sensor.template.publish:
              id: irrigation_controller_status
              state: $zone2 Active
        on_turn_off:
        - then:
          - text_sensor.template.publish:
              id: irrigation_controller_status
              state: Idle
      enable_switch: Enable ${zone2}
      run_duration_number:
        id: zone_2_duration
        name: ${zone2} Duration
        min_value: 1
        max_value: 60
        step: 1
        unit_of_measurement: min
        icon: mdi:timer-outline
        mode: BOX
        restore_value: true
        initial_value: 1
      pump_switch_id: ${devicename}_pump
      valve_switch_id: relay2

################
#  Zone Three  #
################
    - valve_switch:
        id: $zone3_id
        name: $zone3
        on_turn_on:
        - then:
          - text_sensor.template.publish:
              id: irrigation_controller_status
              state: $zone3 Active
        on_turn_off:
        - then:
          - text_sensor.template.publish:
              id: irrigation_controller_status
              state: Idle
      enable_switch: Enable ${zone3}
      run_duration_number:
        id: zone_3_duration
        name: ${zone3} Duration
        min_value: 1
        max_value: 60
        step: 1
        unit_of_measurement: min
        icon: mdi:timer-outline
        mode: BOX
        restore_value: true
        initial_value: 1
      pump_switch_id: ${devicename}_pump
      valve_switch_id: relay3
```