esphome/feature-requests #2258: Add additional microphones to voice_assistant

ha88rgc, 2023-06-01:
**Describe the problem you have/What new integration you would like**
I'm trying to use an i2s microphone array (6+1) https://www.dfrobot.com/product-1976.html - one mono, and three stereo microphones with an ESP32. So far, `voice_assistant` only allows for the creation of one microphone instance. I can currently successfully use either the mono mic or one of the stereo mics. I'd be awesome to be able to declare 1+ microphones for `voice_assistant` as this would allow the use of a real 360 degree microphone array, which is quite important for a voice assistant and is used by Google and Amazon alike in their products I believe.

```
microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: ${i2sDin}
    adc_type: external
    pdm: false
    channel: right
  
  - platform: i2s_audio
    id: echo_microphone2
    i2s_din_pin: ${i2sDin2}
    adc_type: external
    pdm: false
    channel: right

...

voice_assistant:
  microphone: echo_microphone2

```

**Please describe your use case for this integration and alternatives you've tried:**
The use case is **Homeassistant's Year of Voice!!** The DF Robot mic array mentioned above fits the bill perfectly with 6+1 mics and 12 rgb leds for visual feedback. The only thing missing is the `wake word` and more than one microphone!

**Additional context**

nagyrobi, 2023-06-01:
The purpose of a microphone array is a bit more than just having a bunch of microphones connected. The microphones are looking in the same direction and have the same characteristic. They worth nothing by summing up their signals - moreover, phase errors will cause less intelligible signal to the STT system if just dumbly summing them up.

A microphone array works by analyzing the signal incoming from each microphone, and calculating from the phase difference between them, the speaker's relative physical position can be calculated, that's what gives you what marketing materials call _360 degrees microphone_. 

The usage of this information is, for example, to pan/tilt and zoom a camera to a speaker. The audio is actually only used from the center mic (for audio purposes) the signal of the others is only used for position calculations.

Afaik neither HA nor ESPHome can support or use that kind of information yet - to know where the user is speaking from (relative to the microphone).

ha88rgc, 2023-06-01:
@nagyrobi I see. Thanks for clarifying. It would have been nice because the part is packaged nicely haha

nagyrobi, 2023-06-01:
Yes, looks neat indeed. But I'm sure in the future somebody will do it.

snechiporenko, 2023-06-01:
That's what two mics are really for:
[image]

nagyrobi, 2023-06-02:
Acoustic echo cancellation is needed when a full duplex voice call is in progress. Eg. a Skype call or a Teams meeting. So the sound coming out of the speakers when is picked out by the microphones, it's filtered and not fed back to the other side.

stathisktm, 2023-06-13:
Is there any progress?  as I also have such a microphone

Hedda, 2024-08-19:
FYI, stumbled on this new "***voice-kit***" GitHub repository where ESPHome and Home Home Assistant developers from Nabu Casa are developing several new or improved/enhanced audio and media related components, including extended I2S audio for i2c audio microphone and  i2s audio speaker support, including using audio processor (audio DSP hardware) as well as a new native media player with support for FLAC, WAV/PCM, and more advanced resampling, etc. for the upcoming voicekit hardware reference project from Nabu Casa:

- https://github.com/esphome/voice-kit/
  - https://github.com/esphome/voice-kit/tree/dev/esphome/components
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/media_player.py
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/nabu_media_player.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_pipeline.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_reader.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/resampler.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_mixer.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/audio_resampler.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/flac_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/wav_decoder.cpp
      - https://github.com/esphome/voice-kit/blob/dev/esphome/components/nabu/mp3_decoder.cpp
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio
      - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/speaker
      - https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/microphone
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/media_player
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/voice_kit
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/audio_audio_processor
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/voice_assistant
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/speaker
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu_microphone
    - https://github.com/esphome/voice-kit/tree/dev/esphome/components/microphone

Note that I'm only paraphrasing but one of the representatives that is working on that "***voice-kit***"project more or less wrote there that while ESPHome and Home Assistant voice developers from Nabu Casa are now focusing to work on voice assistant features they are still figuring all this around making use of external audio processors and while they are currently only testing the XMOS xCORE chip as a candidate for an ESPHome-based voice-kit reference hardware design for official Home Assistant Voice Assistant development kit they also plan to work on "audio processor" component for ESPHome with hardware-independent architecture that will not be reliant on specific hardware configurations or dependent specifically on the XMOS xCORE DSP chip but instead allow others to add support for additional DSPs as audio processors (i.e. sound co-processors) in the future, (plus the fact that they will make it so that all the I2S settings and pins are still configurable in YAML, meaning that it should at least be possible add support for DSP types to the  "audio processor" component if they work similar to XMOS xCORE DSP chips, as well as different board designs that uses other I2S settings and pins).  That representative also wrote; "_**we will add all the code to the base ESPHome project once things are stable and working well**_".m and noted that ESPHome and Home Assistant / Nabu Casa developers are right now moving very fast and breaking things as they go so working on code for the new voice-kit related components for ESPHome in a separate repository on GitHub here: 

* - https://github.com/esphome/voice-kit/

Anyway, they already added several new experimental features and functions or improvements/enhancements to ESPHome, including a new `nabu_microphone` component that can add better support different microphone channel audio streams.

* https://github.com/esphome/voice-kit/tree/dev/esphome/components/nabu_microphone

and 

* https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio

plus 

  * https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio/microphone

Again, still very early in development so addtional components and more development or changes will sure to come there soon.