esphome/feature-requests #2260: Add function "get_total_sensor" in pulse_counter component

TechEnchanter, 2023-06-02:
**Describe the problem you have/What new integration you would like**

Implement a getter function that allows reading the total pulse count, equivalent to the existing set_total_sensor function.
This getter function will facilitate retrieving the total pulse count, which can be utilized, for example, by a Lambda function for further processing or analysis.
The inclusion of this getter function will enhance the versatility and usability of the pulse_counter component, providing a comprehensive set of functionalities for users.
It will enable users to access and utilize the total pulse count data more effectively in their applications or systems.

**Please describe your use case for this integration and alternatives you've tried:**

Implement periodic storage of total count values (e.g., every 10 minutes)
Enable accurate restoration of total count values at system startup
Ensure synchronized and reliable tracking of gas meter usage
Prevent data discrepancies and loss during power cycles or restarts.

**Additional context**

ssieb, 2023-06-02:
There already is a method for this.  It's the `total:` sensor.

closed 2023-06-02 by ssieb

ssieb, 2023-06-02:
If you need help understanding how to use this, then come to the discord server.