esphome/feature-requests #2261: Add more shapes and type of lines for shapes for display component

facos86, 2023-06-02:
**Describe the problem you have/What new integration you would like**

Hi!

1. Could be very cool to have more shapes (triangles, etc...) in the display component.
2. Also, would very much like to customize the type of lines for the existing shapes (dotted line, etc...)

Regards,

mathieu-mp, 2024-01-14:
PR to add triangle shape, outlined or full filled:
https://github.com/esphome/esphome/pull/6096

bqzero, 2024-04-15:
display component could indeed need further shapes / possibilities to visualize data.
For example a gauge draw option to show current levels like this on lovelace (of course simpler design without the text and arrow)
[image]

nagyrobi, 2024-04-15:
Check https://github.com/esphome/esphome/pull/6363