esphome/feature-requests #2263: Support for AW9523 GPIO Expander and LED Driver

lboue, 2023-06-03:
**Describe the problem you have/What new integration you would like**

No support for AW9523 GPIO Expander and LED Driver in the main branch.

**Please describe your use case for this integration and alternatives you've tried:**

- AW9523 GPIO Expander is used in the [M5Stack CoreS3 ESP32-S3 ](https://docs.m5stack.com/en/core/CoreS3) IoT controller.

I tried this componant:
- Tested with Adafruit AW9523 GPIO Expander and LED Driver Breakout https://www.adafruit.com/product/4886

**Additional context**

[]([Adafruit AW9523 GPIO Expander and LED Driver for Arduino](https://github.com/adafruit/Adafruit_AW9523)
[IO Expander APIs](https://esp-cpp.github.io/espp/io_expander/index.html) » AW9523 I/O Expander
[AW9523B datasheet](https://cdn-shop.adafruit.com/product-files/4886/AW9523+English+Datasheet.pdf))

lboue, 2023-10-18:
I found this componant for ESPhome : https://github.com/jenscski/esphome-components

Tuckie, 2025-04-16:
Just realized that this is used by the M5Stack StamPLC https://docs.m5stack.com/en/core/StamPLC

It would be great to get this expander integrated into the main project.

patrick3399, 2025-05-13:
> Just realized that this is used by the M5Stack StamPLC https://docs.m5stack.com/en/core/StamPLC
> 
> It would be great to get this expander integrated into the main project.

You can try my YAML for StamPLC [Here](https://github.com/patrick3399/esphome_components/blob/main/m5stack-stamplc.yaml)

lboue, 2025-05-13:
@patrick3399 Could you open a [PR](https://github.com/esphome/esphome/pulls) to integrate `aw9523 `?

lboue, 2025-05-14:
I also found this implementation:
https://github.com/jenscski/esphome-components/tree/main/components/aw9523