esphome/feature-requests #2264: Expanding lambda effect examples

JJKing42, 2023-06-04:
**Describe the problem you have/What new integration you would like**

  So i have spent the last two evenings learning to use addressable lights within ESPHome
  And when it comes to light effects, most examples can be copy pasted for a quick and easy test, or just to use those.
  But for the Lambda effect, some people might be scared off, by the aspect of using another language when perhaps being new to 
  YAML as well. 
  So i'm suggesting an expansion into the docs for this specific effect, with a few examples one could actually copy/paste, see it work 
  and tweak some parameters. Essentially lowering the entry level.
  I'll put some example of ideas for examples in the additional context bit.

**Please describe your use case for this integration and alternatives you've tried:**

I'm too stupid to learn two things at once.

**Additional context**

      - addressable_lambda:
          name: "Cycle Colors"
          update_interval: 10ms
          lambda: |-

            // it.size() - Number of LEDs
            // it[num] - Access the LED at index num.

            // Set the LED at num to the given r, g, b values
            // it[num] = Color(r, g, b);

            // Get the color at index num (Color instance)
            // it[num].get();

            static uint16_t num1 = 0;
            static uint16_t red = 255;
            static uint16_t green = 3;
            static uint16_t blue = 3;

            uint16_t num = num1;
            uint16_t numled = 71;

            if(red > 251 && blue < 4 && green < 252){
              green+=5;
  
            } else if (red > 3 && green > 251 && blue < 4){
              red-=5;
  
            } else if (red < 4 && green > 251 && blue < 252){
              blue+=5;
              
            } else if (red < 4 && green > 3 && blue > 251){
              green-=5;
              
            } else if (red < 252 && green < 4 && blue > 251){
              red+=5;
              
            } else if (red > 251 && green < 4 && blue > 3){
              blue-=5;
              
            }

            it[num] = Color(red, green, blue);
            it[(numled) - num] = Color(red, green, blue);
            if(num1 == (numled +1) / 2){
              num1 = 0;
            } else {
              num1++;
            }

      - addressable_lambda:
          name: "ColorScan"
          update_interval: 50ms
          lambda: |-

            // it.size() - Number of LEDs
            // it[num] - Access the LED at index num.

            // Set the LED at num to the given r, g, b values
            // it[num] = Color(r, g, b);

            // Get the color at index num (Color instance)
            // it[num].get();

            static uint16_t num1 = 0;
            static uint16_t red = 255;
            static uint16_t green = 3;
            static uint16_t blue = 3;
            static bool dir = false;
            
            uint16_t num = num1;
            uint16_t numled = 71;

            if(red > 251 && blue < 4 && green < 252){
              green+=5;
  
            } else if (red > 3 && green > 251 && blue < 4){
              red-=5;
  
            } else if (red < 4 && green > 251 && blue < 252){
              blue+=5;
              
            } else if (red < 4 && green > 3 && blue > 251){
              green-=5;
              
            } else if (red < 252 && green < 4 && blue > 251){
              red+=5;
              
            } else if (red > 251 && green < 4 && blue > 3){
              blue-=5;
              
            }
  
            it[num] = Color(red, green, blue);
            if(dir) {
              if(num1 == numled){
                dir = false;
              } else{
                num1++;
              }
            } else  {
              if(num1 == 0){
                dir = true;
              } else{
                num1--;
              }
            }

nagyrobi, 2023-06-04:
Why don't you just sumbit these as PR to the doc?

JJKing42, 2023-06-08:
> Why don't you just sumbit these as PR to the doc?

Because it took me over an hour to be lead here. I got no clue how to navigate github