esphome/feature-requests #2267: Support new date, time and date/time entities in Home Assistant

mill1000, 2023-06-07:
**Describe the problem you have/What new integration you would like**

The latest Home Assistant release (2023.6) just added support for integrations to provide [date, time and date/time entities](https://www.home-assistant.io/blog/2023/06/07/release-20236/#new-entities-date-time-datetime).

Quote from the release announcement.
> These entities allow integrations to provide date, time, or date/time input controls from the UI. They are very similar to the [input datetime](https://www.home-assistant.io/integrations/input_datetime) helper but are available for use by integrations.

I would like to see ESPHome support these entities so we can have UI configuration for time automation.

**Please describe your use case for this integration and alternatives you've tried:**

I currently implement a solution using a [input_datetime](https://www.home-assistant.io/integrations/input_datetime) helper with a [Home Assistant attribute sensor](https://esphome.io/components/sensor/homeassistant.html) to pull in the timestamp and then make comparisons using an `on_time` trigger every minute.

At very least, it would be nice if ESPHome could provide the date, time or date/time entities even if they don't integrate automatically with the `on_time` trigger. This would at least eliminate the need to manually create and link an input_datetime to the device.

**Additional context**

#1333 also made a similar request, but there wasn't HA support at the time. The linked alarm clock example utilizes a similar solution to the one I described above.

jesserockz, 2023-06-08:
I have started working towards this.

mill1000, 2023-11-29:
I don't this feature has been implemented yet.

thargy, 2024-01-17:
@jessrockz Is there any progress on this? We can add an Input Helper and text_sensor to do this, but it's clunky.  The scenario I'm looking at is to allow me to set a timeout for a relay and know it will be honoured even if the HA connection goes down. It seems far better to tell the ESPHome instance when I want the relay to be switched off, rather than trusting that HA will be around to actually turn it off in the future (particularly for longer timeouts).  A datetime sensor on ESPHome that I can set when I turn on the relay would allow the ESPHome code itself to check when the timer has expired and automatically turn off the relay.

Unfocused, 2024-03-20:
`datetime` was just added in v2024.3.0:
- https://esphome.io/components/datetime/
- https://esphome.io/components/datetime/template
 
No `date` or `time` yet.

Unfocused, 2024-03-23:
`time` support is coming up:
- https://github.com/esphome/esphome/pull/6399

Unfocused, 2024-04-16:
Support for the `time` type will be in ESPHome 2024.4.0, which is currently in beta.

The final piece is the `datetime` type (combining both `date` and `time` in one entity), which is being worked on in this PR:

- https://github.com/esphome/esphome/pull/6513

closed 2024-04-25 by jesserockz