esphome/feature-requests #2277: Support for MAX17048 used in many Adafruit Feathers for I2C battery monitoring

circuit69, 2023-06-19:
**Describe the problem you have/What new integration you would like**

Adafruit is using a MAX17048 to monitor the battery status, this uses I2C, a component that supports this chip would be a huge help using these popular dev boards.

**Please describe your use case for this integration and alternatives you've tried:**

The MAX17048 chip is becoming more common on Adafruit boards, showing voltage and battery percent.

**Additional context**

What is the process for creating something like this, is there a guide or tutorial?

max246, 2023-07-13:
Could have a look in the next coming weeks.

JingleheimerSE, 2023-07-18:
Stumbled on this which might be helpful: [https://community.home-assistant.io/t/adafruit-max17048-lipo-battery-gauge-with-esphome/524005/6](https://community.home-assistant.io/t/adafruit-max17048-lipo-battery-gauge-with-esphome/524005/6)

lboue, 2024-03-24:
I am currently working on a external component that supports this MAX17048 chip.

ilikecake, 2025-01-18:
What is the status of this? I recently created a component for the [LC709203F](https://github.com/esphome/esphome/pull/8037) (the other battery monitor that Adafruit uses). I was planning to make a similar component for this when I needed it, but if you already have it done, that would save me some time. :)

JL00001, 2025-01-20:
> What is the status of this? I recently created a component for the [LC709203F](https://github.com/esphome/esphome/pull/8037) (the other battery monitor that Adafruit uses). I was planning to make a [...]

There seems to be one already on github.

https://github.com/Option-Zero/esphome-components/tree/main/components/max17048

sample YAML.
```
external_components:
  - source: github://Option-Zero/esphome-components
    components: [max17048]
sensor:
  - platform: max17048
    battery_voltage:
      name: Battery voltage
      id: batt_v
    battery_level:
      # This uses the SOC (state of charge) value from the MAX17048.
      # Take it with a grain of salt.

      # There is a sophisticated algorithm that takes into account
      # the battery's voltage, current, temperature, and more.
      # However, don't expect it to be too accurate without battery-specific calibration.
      # As the data sheet says (emphasis mine), the chip has 
      # "a preloaded ROM model that performs well for *some* batteries."
      name: Battery level
      id: batt_pct
    rate:
      name: Battery discharge rate
      id: batt_discharge_rate
```

I am not the creator, nor the supporter, nor is this officially maintained by the ESP Github; I am just you a few weeks ahead, but this could be a starting off point for the official MAX17048 component.

The data sheet [LINK](https://www.analog.com/media/en/technical-documentation/data-sheets/max17048-max17049.pdf) describes a config register for different battery chemistry as well as several other features that could be useful.

ilikecake, 2025-01-24:
After further research: Esphome already supports the [MAX17043](https://esphome.io/components/sensor/max17043.html). Comparing the datasheets, the same code and functions should work for this MAX17048. I have tried it with one of the new Adafruit ESP32-S3 Feather boards with the MAX17048 and it appears to work fine.

closed 2026-10-12 by bdraco