esphome/feature-requests #2278: usbip implementation (TCP-to-USB similar to TCP-to-UART)

psychowood, 2023-06-19:
**Describe the problem you have/What new integration you would like**

I'd like to suggest an implementation that could be used to pass usb data over network to connect remote devices, it could be based of the [USB/IP Project](https://usbip.sourceforge.net/), there's even a [PoC](https://github.com/chegewara/esp32-usbip-poc) for ESP32.

**Please describe your use case for this integration and alternatives you've tried:**

My main use case is handling an USB-only UPS put in a place where it doesn't have access to wired network, to bridge with the NUT integration of HA. I suppose there are other scenarios for non-UART USB-only devices out there, like getting stills from a USB webcam.

For my specific use case I could probably add a linux-based sbc to handle NUT, but that would just add another full fledged general-purpose device in the network.

**Additional context**

panayotovip, 2024-12-28:
it is will be nice feature

jankornblum, 2025-02-04:
That would be great.