esphome/feature-requests #2279: Support for ILI6485 and ILI9485 based displays

bipsendk, 2023-06-20:
Support for displays based on ILI6485 and ILI9485 driver chips.

**Describe the problem you have/What new integration you would like**

Enabling support for one or both chips would (hopefully) make it possible to utilize the ESP32-S3 module based displays available on https://www.aliexpress.com/item/1005004788147691.html - which could be awesome for display/control panels

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nielsnl68, 2023-06-20:
Can you search the specifications of this display (the datasheets etc.) or code to set them up.
I'm happy to add them, but i have not the time to search the init strings etc.

bipsendk, 2023-06-20:
For ILI9485 I found this info: https://github.com/Bodmer/TFT_eSPI/discussions/2336
I assume the same restrictions apply on ESP32 ?
I will continue to check for data on ILI6485  - otherwise the vender of the module has some example arduino code - not sure if that can help ?

nielsnl68, 2023-06-20:
Ha, sadly as *Bodmer* said on his repo "That display has an RGB interface" and at this moment of time espHome does not support  this kind of communications yet,

bipsendk, 2023-06-21:
This is what I could find on ILI6485 so far:

[image]

Maybe this is also RGB interface... :-/

MiggyMan, 2023-11-14:
Is this likely to ever be supported ?

I'm working on a "universal remote" front end display project atm and as it's stands between esphome and openhasp as the brains for this and while openhasp does have the better ui options esphome looks like it will let me configure mqtt in a much more user friendly manner as far as the resulting nodered code that does the leg work does and also since i want this to be battery powered deep sleep is very much a wanted/needed feature which again only esphome presents at present.

andreaschizzali, 2025-02-19:
I believe thet the driver ILI9485 does not exist.
I have a similar display model ESP32-487S043 with the same problem. 
In the documentation from Sunton about the display module there are interesting documents.
In the specifications of the screen (_4.3” 480X272-TN bare screen specs.pdf_), the company Shenzen Jincai Intelligent Co., LTD states: "Driver IC  ILI6485". In the document _ESP32-4827S043 Specifications-EN.pdf_ the same company states "Driver IC  ILI6485" in the document ESP32-4827S043 Specifications-EN.pdf. 
It seems to me there is merely a simple transcription error, so the effective driver chip is the ILI6485.
The datasheet for the ILI6485 can be easily found (https://aitendo3.sakura.ne.jp/aitendo_data/product_img/lcd/tft2/QT43272B576/ILI6485_DTS_V001_20180403.pdf). The informations from the datasheet correspond to the values emploied in the demo programs. 
I hope that helps.

xvart, 2025-05-10:
there is obviously some form of driver as the ESP32-487S043 does display something, sure it may be using one core to drive the display but there is some form of driver just not the usual SPI thing and I guess the ESP32 holds the displays frame buffer in SPIRAM