esphome/feature-requests #2281: sync binary sensor state to light or fan on boot

uddinkayyum, 2023-06-20:
**Describe the problem you have/What new integration you would like**

There should be a category for momentary and wall_switch in binary_sensor, so user no need to write extra script to sync light or fan to binary_sensor on boot.

**Please describe your use case for this integration and alternatives you've tried:**

Tasmota already have support for momentry switch, currently in esphome, on_press does not trigger on boot so light or fan does not turn on after power cut. 

Here is the configuration currently using for solve this issue:
```
esphome:
    ...
    on_boot:
    priority: 600
    then:
      - script.execute: sync_on_boot

binary_sensor:
  - platform: gpio
    ...
    on_press:
      then:
        - light.turn_on: light1
    on_release:
      then:
        - light.turn_off: light1

  - platform: gpio
    ...
    on_press:
      then:
        - fan.turn_on: fan1
    on_release:
      then:
        - fan.turn_off: fan1

 script:
  - id: sync_on_boot
    then:
      - script.execute: sync_switch1
      - script.execute: sync_switch2

  - id: sync_switch1
    then:
      - wait_until:
          condition:
            binary_sensor.is_on: switch1
          timeout: 1s
      - if:
          condition:
            binary_sensor.is_on: switch1
          then:
            - light.turn_on: light1
 
  - id: sync_switch2
    then:
      - wait_until:
          condition:
            binary_sensor.is_on: switch2
          timeout: 1s
      - if:
          condition:
            binary_sensor.is_on: switch2
          then:
            - fan.turn_on: fan1
```
**Additional context**

Maybe esphome should had something like this:
```
binary_sensor:
  - platform: gpio
     ...
     sync_to_device:
        - id: light1
```

ssieb, 2023-06-20:
`publish_initial_state: true`

closed 2023-06-20 by ssieb

uddinkayyum, 2023-06-20:
> `publish_initial_state: true`

Thanks, but there is no clear info about publish_initial_state: true in docs, at least add they also call on_press and on_release on boot. Thats why i confused and request for feature.

nagyrobi, 2023-06-21:
See https://esphome.io/components/binary_sensor/index.html?highlight=publish_initial_state#base-binary-sensor-configuration