esphome/feature-requests #2282: PCF8574 interrupt pin

pantherale0, 2023-06-21:
**Describe the problem you have/What new integration you would like**
I have a PCF8575 that is currently used purely for inputs from momentary buttons, software polling seems extremely slow and doesn't pick up state changed very quickly. This FR is to request support for the interrupt pin of a PCF8575 to allow near instant updates on events.

**Please describe your use case for this integration and alternatives you've tried:**
Improvement to existing integration, no alternatives available due to limited configuration options

**Additional context**

pantherale0, 2023-06-21:
This is the specific board I'm using https://amzn.eu/d/1fTJjcL

mstuczko, 2024-03-31:
Adding support for interrupts would open possibility to use pule counters....
Support for interrupts is very much needed (and overdue).

bartowl, 2025-10-06:
I know this is quite old, but I upvote this. Adding suport for interrupt pin is not that complicated and would allow asynchronous notifications on input change instead of polling which is not optimal nor reliable

4b686f61-2c1b, 2026-04-16:
Now it's time to add the RV-3032-C7 RTC clock. Although ESPhome has the [RX8130](https://esphome.io/components/time/rx8130/) Which comes in a pretty small form factor, It's not temperature compensated.

pantherale0, 2026-04-16:
> Ok, sorry for the noise, but I've been able to trace it down.
> 
> The following situation causes a boot loop with above stack trace:
> * pcf8574 with configured input interrupt
> * cable connection from pcf interrupt to configured GPIO
> * light with `platform: esp32_rmt_led_strip`
> 
> Here is a minimal non-working yaml file:
> ```
> esphome:
>   name: esp-keller
>   # Prefix für Entitäten: esp_keller_
>   friendly_name: ESP Keller
> 
> 
> esp32:
>   board: az-delivery-devkit-v4
>   framework:
>     type: esp-idf
> 
> 
> # Enable logging
> logger:
>   level: DEBUG
>   #level: INFO
>   logs:
>     component: ERROR
>     sml: INFO
>     #sensor: INFO
> 
> 
> api:
> 
> 
> ota:
>   - platform: esphome
>     password: emohpse
> 
> 
> safe_mode:
>   disabled: true
> 
> 
> wifi:
>   ssid: wlan
>   password: nalw
>   use_address: 192.168.20.4
> 
> 
> i2c:
>   - id: i2c_pcf
>     sda:
>       number: 4 # Grün
>     scl:
>       number: 2 # Blau
>       ignore_strapping_warning: true # connected to on-board LED, must be left floating or LOW to enter flashing mode
>     scan: false
>     frequency: 100kHz
> 
> 
> pcf8574:
>   - id: pcf8574_2
>     i2c_id: i2c_pcf
>     address: 0x21
>     pcf8575: false
>     interrupt_pin:
>       number: 39
>       mode:
>         input: true
> 
> 
> light:
>   - id: light_sleeping_crown
>     name: Schlafzimmer Stuckleiste
>     platform: esp32_rmt_led_strip
>     rgb_order: RGB
>     is_wrgb: true
>     chipset: WS2811
>     pin:
>       number: 14
>     num_leds: 166
> ```

Better to report this as a new bug than here in the feature requests :)