esphome/feature-requests #2285: Support IRK for BLE tracking

d0rh4ns, 2023-06-23:
**Describe the problem you have/What new integration you would like**

Introduce IRK support for the already existing BLE tracking functions to support Apple devices, which change MAC addresses regularly. Current methods of identification provided by ESPHome don't work in those cases. Identifying a device by its IRK would solve this. 

**Please describe your use case for this integration and alternatives you've tried:**

Tracking Apple devices with the BLE tracking functions of ESPHome. ESPresense does this, but requires separate ESP32, which could be avoided if this feature is implemented in ESPHome.

**Additional context**

pwntester, 2024-03-21:
This seems to be implemented now, thanks @chbmuc! Any chances you can extend the support to `ble_rssi` platform for irk devices?

chbmuc, 2024-03-22:
I think it should be easy as the code looks quite similar. I will try to do this over the weekend.

mano3m, 2025-12-30:
This issue can be closed?