esphome/feature-requests #2286: Support other MPP Inverters than PIP

JakobTewes, 2023-06-23:
**Support other MPP Inverters than PIP**

**Get other MPP or subbrand inverters running:**

Mat931, 2023-06-25:
Which inverter are you looking for? PIP isn't the only one supported right now.

 - Growatt:
   https://esphome.io/components/sensor/growatt_solar.html

 - Havells:
   https://esphome.io/components/sensor/havells_solar.html
 
 - PipSolar:
   https://esphome.io/components/pipsolar.html

 - Victron (via external component):
   https://github.com/Fabian-Schmidt/esphome-victron_ble

 - EPEVER (via Modbus component):
   https://devices.esphome.io/devices/epever_mptt_tracer_an

 - SUN-1000G2/SUN-2000G2:
   via Modbus component or my pull request:
   https://github.com/esphome/esphome/pull/4958

 - Every other inverter with a Modbus (RS232/RS485) interface

You can also check if your inverter is supported by HomeAssistant (without ESPHome)

JakobTewes, 2023-06-25:
Dammn, sorry for that - was misleading and I’ve not even noticed ;-).
I ment other MPP/EASun type inverters with slightly different protocol. 

With python module mpp-solar, one can define the protocol by using the option „-p“ and then define the corresponding lingo for the inverter

Mat931, 2023-06-25:
Do you know which protocol your inverter is using? [Here](https://github.com/jblance/mpp-solar/tree/master/docs#protocol--command-references) is a list of protocols. (the files starting with `PI`)

Did you already test your inverter to make sure it isn't already supported by ESPHome? In the [documentation](https://esphome.io/components/pipsolar.html#overview) it says that a wide variety of inverters can be used with the pipsolar component

If yours isn't supported you can check the protocol datasheet and modify the [pipsolar](https://github.com/esphome/esphome/tree/dev/esphome/components/pipsolar) component. You can ask for help on the [Discord server](https://discord.gg/KhAMKrd).

Saentist, 2023-07-08:
Protocol Is **Voltronic** but somebody rename it as **PIPSOLAR**, 
but there is subversions for different model series.