esphome/feature-requests #2287: Sprinkler component to be compatible with H Bridge for Solenoid Valves

raidnet-ms, 2023-06-23:
**Describe the problem you have/What new integration you would like**

There are two types of solenoid valves in the market that could be used to manage sprinklers. Normally Open and bistable solenoids. 

Since the first type can be easy managed with sprinkler component that uses switch to drive the relay wich open and closes the solenoid, **bistable solenoids** need to reverse their polarity to change their state, so the **HBridge component** would suit better for them.

**Please describe your use case for this integration and alternatives you've tried:**

Unfortunately the sprinkler component can't manage outputs in place of switch and you need to complicate your yaml config to make them work.
Of course there is a way to make them work with actual switch but if you could implement the use of outputs this could be managed with less configuration moreover gpio output component has the vantage of being hidden by default in the frontend so you won't have a duplicated switch for each sprinker.

I imagine something like this:

```
output:
  - platform: sprinkler
    id: valve_A_on
    pin: GPIO5
  - platform: sprinkler
    id: valve_A_off
    pin: GPIO4

sprinkler:
  - platform: hbridge
    id: garden_valve
    name: "Garden Valve"
    pin_a: valve_A_on
    pin_b: valve_A_off
```

**Additional context**

Bistable solenoids are necessary when you plan to use a solar-battery powered sprinkler controller because those solenoids do not consume energy during the irrigation process. They change state in 800ms then they keep the state until you provide another short signal with reversed polarity and to manage this H-bridge would be the best approach.

With bistable solenoids if during the irrigation process there’s a power interruption, the water would keep flowing, but Hbrighe has the perfect solution that can manage this problem:

```
on_boot:
  then:
    - sprinkler.hbridge.brake: all_valves
```

nagyrobi, 2023-06-28:
@kbx81

kbx81, 2023-06-29:
I am aware of the (binary) output component and H-bridge — the motivation for using `switch` components was that some people _want_ to manually control valves from the front end, particularly when testing their system/config. Using the `output` component would effectively make this impossible (or, at least, substantially more difficult/messy). The option that keeps the most people happy is to let the user choose whether the switch appears in the front end, which is easy to do with a `switch` component. 😇 (I would also argue that, conceptually, the switch component is somewhat easier to understand because you can “see” it — the output component is more of an abstraction which is more difficult for some people to grok.) In short, I see no benefit of changing it to use an `output` component as opposed to a `switch`.

As for control of an H-bridge driver IC, the physical implementation varies from chip to chip. Some have a single pin to drive the polarity of each output; others have an "enable" pin and a "direction" or "polarity" pin, which won't play as nicely with the existing implementation of `sprinkler`.

An example of the former is the [L9110](https://cdn-shop.adafruit.com/product-files/4489/4489_datasheet-l9110.pdf) available from [Adafruit](https://www.adafruit.com/product/4489). Based on the truth table at the top of the second page of the PDF, I expect this IC (or anything similar to it) would work perfectly with the existing implementation of `sprinkler`. There is no need whatsoever for an `output` component to accomplish this; switches will work just fine and no additional configuration is necessary to achieve the correct behavior. To implement this physically, you'd wire `IA` to one GPIO pin and `IB` to a different GPIO pin. Then, in the `sprinkler` valve configuration, one of those GPIOs would become the `valve_off_switch_id` and the other the `valve_on_switch_id`. Done.

In short, the type of H-bridge IC that you choose to use is going to determine whether or not `sprinkler` can directly control your valves with its switches. If you use an IC with a more complex control interface, then, yes, the yaml configuration is going to become messy. In general, for sake of simplicity, I recommend sticking with an IC similar to the `L9110` I mentioned above.

nagyrobi, 2023-06-30:
> The option that keeps the most people happy is to let the user choose whether the switch appears in the front end, which is easy to do with a `switch` component.

Additionally, one could use not only GPIO-based relays, but also other relay modules or switches which connect though different ways, like I2C or ModBus cards.