esphome/feature-requests #2296: Temperature compensation for Ultrasonic Component

aura6174, 2023-06-29:
**Describe the problem you have/What new integration you would like**

Ultrasonic sensor are quite sensitive to environment temperature, in reality when temperature delta is huge distance shift by +-10cm. It's is more ideal when ultrasonic have temperature compentation feature:
[image]
**Please describe your use case for this integration and alternatives you've tried:**

Current config to resolve this issue
```
  - platform: ultrasonic
    ...
    id: ping_sensor
    filters:
      - lambda: |-
          return x / 343.0 * (331.3 + 0.606 * id(temp_sensor).state);
```
**Additional context**

```
 - platform: dallas
    address: 0x17020b9177583c28
    name: "Internal Temperature"
    resolution: 12
    id: temp_sensor

  - platform: ultrasonic
    trigger_pin: D5
    echo_pin: D6
    name: "Ping Sensor"
    update_interval: 100ms
    accuracy_decimals: 3
    timeout: 1.5m
    pulse_time: 20us
    id: ping_sensor
    compentation_source: temp_sensor
```

aura6174, 2023-06-29:
okay, this feature was an optional improvement rather than force implement

drahdiwaberl, 2023-07-25:
thanks for sharing, though