esphome/feature-requests #2298: Range filter for sensor component

aura6174, 2023-06-29:
**Describe the problem you have/What new integration you would like**

Simply ignore value if sensor publish out of range value.
**Please describe your use case for this integration and alternatives you've tried:**

```
- platform: sensor
  name: "flow meter"
  ...
   filters:
    - lambda: |-
          if (x > -20 && x < 60) {
            return x;
          } else {
            return 0;
          }
```
**Additional context**

```
   filters:
      - range:
           - above: -20
           - below: 60
```

bkaufx, 2023-06-30:
The lambda should work if you just return {} instead of 0.  https://esphome.io/components/sensor/index.html#lambda

> To prevent values from being published, return {}:

aura6174, 2023-06-30:
you miss one thing this one is **flow meter** where value goes +40 when water filling tank and -10 when drown water from tap, so 0 is more ideal than {}.

closed 2023-10-25 by jesserockz