esphome/feature-requests #2300: TM1638: Add support for 10 digit common anode displays

Qeteshpony, 2023-06-30:
**Describe the problem you have/What new integration you would like**

I have a clock using a TM1638 to drive 10 7-segment digits to display time and date. I would like to be able to use this with ESPHome. This is a usecase also described in the ICs datasheet so it would add to the support for this cheap chip.

**Please describe your use case for this integration and alternatives you've tried:**

It would be possible to use devices that have a 10 digit display with ESPHome. I got it running with arduino through a lot of trial and error but I was not able to find out how to do the same for ESPHome.

**Additional context**

I adapted a driver for arduino a while ago to support this usecase. The code for that can be found at https://gist.github.com/Qeteshpony/7509e7c3c2acfd19f5fb7bdb90daae49

nagyrobi, 2023-07-01:
Did you look at https://esphome.io/components/display/tm1638.html ?

Qeteshpony, 2023-07-01:
> Did you look at https://esphome.io/components/display/tm1638.html ? 

Yes but I found no way to switch to 10 digit common anode mode since the digits need to be sent in a different coding to the IC - basically the whole display needs to be "turned" by 90° because the SEG and GRID pins are switched. The way that has to be done can be seen in the linked .ino file at the end

> Do you have a link to the product?

Not really since the one I use here is a DIY project. But since this mode of usage is described in the datasheet it should be considered a standard function of the chip.

nagyrobi, 2023-07-01:
Well you should try to modify the original ESPHome component, and use it as custom component. And eventually submit a PR.

Qeteshpony, 2023-07-01:
> Well you should try to modify the original ESPHome component, and use it as custom component.

I have looked into that but I wasnt able to find out how to create a custom display driver that can be used with ESPHome. Unfortunately the examples for custom components go nowhere near that and I never wrote object based C-code before.