esphome/feature-requests #2304: Serve www.js from dashboard to allow local hosting

xrxca, 2023-07-03:
**Describe the problem you have/What new integration you would like**

It would be nice if the www.js was served from the esphome dashboard, so the web page could be handled locally without adding the www.js to the device firmware, and without another local server.

**Please describe your use case for this integration and alternatives you've tried:**

I run the esphome docker container, and since it's always running, it is a logical location for local serving of the url_js: for the web component.

This could be done either by adding it to the dashboards static folder and using a url such as `http://192.168.100.100:6052/static/www.js`, or a handler added to the tornado.web.Application config to statically serve the jsfile from the config directory or a config/web directory. 

**Additional context**

I'm currently building a custom docker image with www.js injected into the static folder and using the static/www.js url.

ohadlevy, 2023-08-03:
if its running as part of esphome home assistant addon, this could be served by HA?

EternityForest, 2023-08-19:
What about gzip encoding the files? If they were only 25% as big, they would fit on flash most of the time, so you wouldn't have a single point of failure, and pretty much all browsers support it.

It would also give a bit more room for adding features to the web later.