esphome/feature-requests #2305: RF 433MHZ & IR Remote as a use for HA connected Device control

hiteshkher, 2023-07-04:
**I would like  RF 433MHZ & IR Remote as a use for HA connected Device control,  (esphome base esp8266/esp32)
 so basically I want yaml file for understand and read RF or IR remoter data,  and give a on esp thru RF Receiver/IR receiver and control esp connected device, I already much search about one esphome, home assistent, and google but not any source gat so please add this features  , Thank you**

**I want use for control device from remote because some time from mobile control not easy**

****

skachati, 2023-10-14:
captive_portal:
#-------------------------------------IR + RF RECEIVER-------------------------
remote_receiver:
  - id: rf_rx
    pin: 1
    dump:
    - rc_switch
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: ir_rx
    pin:
      number: 5
      inverted: true
      mode:
        input: true
        pullup: true
    dump: all

closed 2023-10-14 by nagyrobi

hiteshkher, 2023-10-14:
remote_receiver:
  pin: D6
  dump:
   - rc_switch

I am using this RF RX, working is good but 100% not satisfied, because sometimes RF signal not work, so i want some time press the button 2 or 3 time then work maybe communication issues

hiteshkher, 2023-10-14:
filter: 250us
idle: 4ms
buffer_size: 2kb
This thing some time frequently false data, and hang the system
So i use only 
remote_receiver:
pin: D6
dump:

rc_switch