esphome/feature-requests #2307: Feature request: Aklarm Control Panel

hwstar, 2023-07-06:
**Describe the problem you have/What new integration you would like**

1. Add per-zone entry delay attribute so that some zones would remain as instant, (example: glass break detectors)  while others would would trigger the pending timer (example: front door)
2. For the alarm control panel during the time it is in the pending state, add an attribute to disable certain zones set to instant alarm so that the alarm isn't triggered by the PIR's due to people walking to the disarm point. 

**Please describe your use case for this integration and alternatives you've tried:**

1. A user comes home, and enters through the front door. The alarm state goes to pending. Unbeknownst to the user, a home invasion burglar is waiting for the user to come home, they then break a window covered by a glass break sensor to enter just after a user enters through the front door. In the existing code there would be no alarm as all zones go through the pending timer before triggering the alarm.
2. When a user arrives home Passive Infrared Sensors could be triggered when the user makes their way to the disarm point. If these are set to instant per use case #1, then the alarm is going to get triggered inadvertently. However if an attribute is set preventing triggers in the pending state for the PIR sensors which will get triggered then the alarm can be disarmed before the pending state timer expires. 

**Additional context**

This feature is implemented on the Ademco/Honeywell Vista 20 panel

nagyrobi, 2023-07-06:
@grahambrown11

grahambrown11, 2023-07-10:
Sure will add it to the feature list...
The aim for 1st implementation was to get a simple panel with all the moving parts for the integration into HA, then build out more advance stuff...
You can still achieve your use case with the existing code by triggering the alarm from the binary sensor automation actions, more config but is still possible