esphome/feature-requests #2311: tuya_light: restore light state from Tuya MCU instead of ESP flash memory

kswt, 2023-07-09:
**Describe the problem you have/What new integration you would like**

On Tuya-MCU enabled devices, the second MCU is usually responsible for saving and restoring light state. I suggest to add option to restore light state from it instead of setting some deafult value (restore_mode: ALWAYS_ON ...) or restoring from ESP flash (restore_mode: RESTORE_DEFAULT_ON ...).

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

kswt, 2023-07-09:
Possible solution: https://github.com/esphome/esphome/compare/dev...kswt:esphome:feat/light_restore_from_tuya
It works well on my device, but does not handle RESTORE_INVERTED_... cases.

dlushni, 2026-09-30:
Got tired of waiting for a solution so decided to try a workaround instead. 
Here Tuya MCU light related datapoints are controlled with "number" sliders that don't have any restore logic, these are declared as internal and hidden from HASS UI. On top of that there is now a template light component (exposed to HASS) that controls the sliders and can sync with them if changes are made with the 433MHz remote. Not the solution I was hoping for, but it seems to work OK.

```
esphome:
  name: office-fan
  friendly_name: Office Fan
  on_boot:
    priority: -100
    then:
      - delay: 100ms
      - globals.set: { id: block_outbound, value: 'false' }
...

globals:
  - id: block_outbound
    type: bool
    initial_value: 'true'

tuya:

# ---------------------------------------------------------
# 1. Tuya Fan works fine with NO_RESTORE
# ---------------------------------------------------------
fan:
  - platform: tuya
    name: Fan
    restore_mode: NO_RESTORE
    switch_datapoint: 1
    direction_datapoint: 8
    speed_datapoint: 3
    speed_count: 6

# ---------------------------------------------------------
# 2. Light component redefined to respect MCU boot state
# ---------------------------------------------------------
output:
  - platform: template
    id: out_bright
    type: float
    write_action:
      - lambda: |-
          if (id(block_outbound)) return;

          if (state == 0.0f) {
            if (id(tuya_switch).state) id(tuya_switch).turn_off();
          } else {
            if (!id(tuya_switch).state) id(tuya_switch).turn_on();

            float target = std::max(2.0f, state * 100.0f);
            if (std::abs(id(tuya_bright).state - target) > 1.0f) {
              id(tuya_bright).make_call().set_value(target).perform();
            }
          }

  - platform: template
    id: out_cct
    type: float
    write_action:
      - lambda: |-
          if (id(block_outbound)) return;

          float target = (1.0f - state) * 100.0f;
          if (std::abs(id(tuya_cct).state - target) > 1.0f) {
            id(tuya_cct).make_call().set_value(target).perform();
          }

light:
  - platform: color_temperature
    id: fan_light
    name: "Light"
    brightness: out_bright
    color_temperature: out_cct
    cold_white_color_temperature: 6000K
    warm_white_color_temperature: 2700K
    gamma_correct: 0
    default_transition_length: 0s
    restore_mode: ALWAYS_OFF

switch:
  - platform: tuya
    id: tuya_switch
    switch_datapoint: 15
    internal: true
    on_turn_on:
      - light.control: { id: fan_light, state: true }
    on_turn_off:
      - light.control: { id: fan_light, state: false }

number:
  - platform: tuya
    id: tuya_bright
    number_datapoint: 16
    internal: true
    min_value: 0
    max_value: 100
    step: 1
    on_value:
      - light.control: { id: fan_light, brightness: !lambda 'return std::max(0.01f, x / 100.0f);' }

  - platform: tuya
    id: tuya_cct
    number_datapoint: 17
    internal: true
    min_value: 0
    max_value: 100
    step: 1
    on_value:
      # Maps inverted 0-100 Tuya CCT to Mireds (166-370)
      - light.control: { id: fan_light, color_temperature: !lambda 'return 370.37f - ((x / 100.0f) * (370.37 - 166.7));' }
```