esphome/feature-requests #2312: In Springler queue cycle is impossible springler wait to be greater than the time the valve is open.

hiperiondev, 2023-07-09:
**Describe the problem you have/What new integration you would like**

I need 3 valves and the pump to activate for 5 minutes with a 10-minute wait between ignitions.
I have added the valves to a queue and when I start the loop it works fine with a wait for valve_open_delay between ignitions.
The problem is that valve_open_delay must be less than run_duration, which makes it impossible for the cycle wait to be greater than the time the valve is open.
The water is taken from a cistern that takes 10 minutes to fill completely and the irrigation empties the cistern in about 5 min

**Please describe your use case for this integration and alternatives you've tried:**

I have added valves to queue and start with sprinkler.start_from_queue

**Additional context**

nagyrobi, 2023-07-10:
@kbx81

kbx81, 2023-07-10:
The purpose of `valve_open_delay` is to mitigate physical issues with valves, such as banging pipes or valves not closing; it is not intended to be used to create extended delays within cycles. The sprinkler controller's internal state machine reflects this and it is the reason that the delay cannot be longer than the valve run duration (the two are coupled/related).

That said, your request could be easily solved with a script:

script:
  - sprinkler.start_single_valve:
      valve_number: 0
      run_duration: 600s
  - delay: 15min
  - sprinkler.start_single_valve:
      valve_number: 1
      run_duration: 600s
  - delay: 15min
  - sprinkler.start_single_valve:
      valve_number: 2
      run_duration: 600s