esphome/feature-requests #2314: Technoline WS2300-15 Wind Speed/Direction Sensor

drobishman, 2023-07-12:
**Describe the problem you have/What new integration you would like**
I noticed that ESPHome only supports the TX20/TX23 anemometer, which are very hard to find. I have a Tehnoline WS2300-15 and unfortunately I cannot use it. I would like to be able to use this sensor in my weather station project.

**Please describe your use case for this integration and alternatives you've tried:**
At the moment I use it with tasmota so that I can integrate it into my project but I want to have everything in ESPHome.
I tried to use it with TX20/TX23 but I don't receive data, as far as I know it has a different datagram.

**Additional context**
I found some information regarding the datagram, I hope it can be helpful: http://www.sdpro.eu/jm/images/allegati/WS2300-15.pdf 
Thanks in advance!

nagyrobi, 2023-07-12:
Ran into the same issue, and thus I bought pr-3000-fsjt-n01 and pr-3000-fxjt-n01 wind speed and direction sensors. I chose the MODBUS (RS485) version. 

Did not have time to implement it with ESPHome but I definitely intend to.

SeByDocKy, 2023-07-13:
> pr-3000-fsjt-n01

There are a lot of Modbus 4S485 wind sensor. Most got slight diference in their protocol. Please find  a ESPhome model with such modbus sensor: https://github.com/SeByDocKy/myESPhome/blob/main/code/weather-station..yaml

drobishman, 2023-07-19:
I think I will order as you suggest, anemometer and wind direction via MODBUS RS485, it seems to me a very good alternative. I found something on Aliexpress and was thinking of buying it: https://www.aliexpress.com/item/1005005500304078.html
Do these devices give relevant data? I'm thinking of controlling the closing of an awning depending on this sensor. WS2300-15 with tasmota does not always transmit relevant data.

Marcin685, 2023-10-04:
Hello, 
Can I find a program for ESPHOME somewhere to read data from the wind speed sensor?

canaula, 2023-12-23:
Hello! 
 I am in the same situation, lost.  
Has anyone implemented this wind sensor with "esphome"?

mhansel0970, 2024-01-19:
Hi,
I have a Technoline WS2300-15, too. I built an arduino uno a few years ago to read out the wind sensor (look at the link in this thread at the beginning) and showing the data on a 20x4 led display. 
Meanwhile I tried to adapt the code of the tx20 implementation (tx20.h) so that it fits for the WS2300-15, but I can't get the program to run.
I am therefore also looking for someone who has implemented this in esphome!

nagyrobi, 2024-05-19:
OK, the wind speed meter on Aliexpress (black plastic), type **pr-3000-fsjt-n01** works with this config, using an RS485/TTL adapter:

```yaml
uart:
  rx_pin: GPIOXX
  tx_pin: GPIOXX
  baud_rate: 4800

modbus:
  id: modbus1

modbus_controller:
- id: wind_speed_meter
  address: 1
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 5s #or whatever you need

sensor:
- platform: modbus_controller
  modbus_controller_id: wind_speed_meter
  name: "Wind speed"
  device_class: wind_speed
  id: wind_speed
  register_type: read
  address: 1
  unit_of_measurement: "m/s"
  value_type: U_WORD
  accuracy_decimals: 1
  filters:
    - multiply: 0.1
```

nagyrobi, 2024-05-19:
The seller offers a helper application for Windows, called `485 Parameter Configuration Tool V5.0.7.1.exe`. The sensor can be connected to the PC with a USB-to-RS485 adapter, and the configuration tool makes it easily possible to change the modbus address to something else, eg. `2` (just type it in the _Addr_ box and press _Setup_ button).

After that it becomes possible to simply connect the sensors in parrallel on the same cable, to a single RS485-TTL adapter attached to a single UART on the ESP:

```yaml
uart:
  rx_pin: GPIOXX
  tx_pin: GPIOXX
  baud_rate: 4800

modbus:
  id: modbus1

modbus_controller:
- id: wind_speed_meter
  address: 1
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 30s #, but updating from interval instead

- id: wind_direction_meter
  address: 2
  modbus_id: modbus1
  setup_priority: -10
  command_throttle: 200ms
  update_interval: 30s #, but updating from interval instead

sensor:
- platform: modbus_controller
  modbus_controller_id: wind_speed_meter
  name: Szélsebesség
  device_class: wind_speed
  id: wind_speed
  register_type: read
  address: 1
  unit_of_measurement: "m/s"
  value_type: U_WORD
  accuracy_decimals: 1
  filters:
    - multiply: 0.1

- platform: modbus_controller
  modbus_controller_id: wind_direction_meter
  name: "Wind direction gear"
  icon: mdi:compass-rose
  register_type: read
  address: 1
  value_type: U_WORD
  accuracy_decimals: 0

- platform: modbus_controller
  modbus_controller_id: wind_direction_meter
  name: "Wind direction degree"
  icon: mdi:compass-rose
  register_type: read
  address: 2
  value_type: U_WORD
  accuracy_decimals: 0
  unit_of_measurement: "°"
```

nagyrobi, 2024-05-19:
@drobishman for **pr-3000-fsjt-n01** and **pr-3000-fxjt-n01** I posted my configs here: https://github.com/esphome/feature-requests/issues/1103#issuecomment-2119325838

nagyrobi, 2024-05-25:
Added to the documentation:
https://devices.esphome.io/devices/Renke-RS-FSJT-N01-Wind-Speed
https://devices.esphome.io/devices/Renke-RS-FXJT-N01-Wind-Direction

Both can be purchased on Aliexpress and similar.

RudikHotVaillant, 2024-07-02:
I bought the zts-3000-fsjt-n01. Also modbus neasuring upto 30 m/s.
Anyone knows what is the difference with the pr-3000-fsjt-n01?

As far as i understand the 3000 stands for the range upto 30 m/s.
The "s" in "fsjt" stands for speed.
The n01 stand for modbus rs485.
The "pr" and "zts" i don't understand.

pr-3000-fsjt-n01 a

BliXem1, 2025-02-21:
Does anyone know if I can get this working with ESPhome? See: https://www.dfrobot.com/product-1308.html.

leonundjulie, 2025-07-24:
Hello. I've also just purchased the two sensors **Renke-RS-FSJT-N01-Wind-Speed** and **Renke-RS-FXJT-N01-Wind-Direction]** for measuring wind speed and direction. I'm currently gathering information.

First, I contacted the manufacturer to request the Windows tool for changing the address. Does anyone use this tool? With the version I received by mail, my virus scanner immediately kicks in :-(

Ultimately, I want to use the sensors to receive the weather data in my iOBroker (it supports Modbus and MQTT). So, my approach is to set up an interface with TASMOTO (I'm at least a little familiar with it, as I have several in use). So, I have the following questions:
- Does anyone use the sensors with iOBroker?
- Does anyone use the sensors with a TASMOTA?
- Does TASMOTA support the RS485 protocol?

SeByDocKy, 2025-07-24:
> Hello. I've also just purchased the two sensors **Renke-RS-FSJT-N01-Wind-Speed** and **Renke-RS-FXJT-N01-Wind-Direction]** for measuring wind speed and direction. I'm currently gathering [...]

You have already an ESPhome support here : https://devices.esphome.io/devices/Renke-RS-FSJT-N01-Wind-Speed

leonundjulie, 2025-07-24:
I know that page. But how to incorporate that into my idea using TASMOTA on ESP8266?

SeByDocKy, 2025-07-24:
> I know that page. But how to incorporate that into my idea using TASMOTA on ESP8266?

Sorry but you are in the ESPhome git.... so I don't know how to make modbus RTU over RS485 with tasmota.... Probably it's doable. Seems only two registers to read... sound not complex at all.

leonundjulie, 2025-07-25:
I am aware about the Headline of this forum, of course. However, because this is a topic with nearly nothing in the www, I hoped to find the initializing idea in this group ;-)

SeByDocKy, 2025-07-25:
> I am aware about the Headline of this forum, of course. However, because this is a topic with nearly nothing in the www, I hoped to find the initializing idea in this group ;-)

I don't know why you need tasmota.... to make MQTT ? ... In ESPhome you can activate MQTT as well...