esphome/feature-requests #2318: In sprinkler controller is not possible to test if a valve is disabled

hiperiondev, 2023-07-14:
**Describe the problem you have/What new integration you would like**

In sprinkler controller is not possible to test if a valve is disabled.

**Please describe your use case for this integration and alternatives you've tried:**

I can enable/disable valve via valve_switch_id state but there aren't any method to test this state.

**Additional context**

nagyrobi, 2023-07-14:
@kbx81

kbx81, 2023-07-14:
> In sprinkler controller is not possible to test if a valve is disabled.

It absolutely is possible. As described in [the documentation here](https://esphome.io/components/sprinkler.html#extending-the-sprinkler-controller-s-switches), the switch components the `sprinkler` component generates are based on ESPHome's `switch` core; you just need to extend their configuration in the yaml to include an `id` as shown in the examples in the documentation. As such, all of the typical [actions](https://esphome.io/components/switch/index.html#switch-toggle-action), [conditions](https://esphome.io/components/switch/index.html#switch-is-on-switch-is-off-condition) and [triggers](https://esphome.io/components/switch/index.html#switch-on-turn-on-off-trigger) work with them. They also work just as any other switch when used in [lambdas](https://esphome.io/components/switch/index.html#lambda-calls).

closed 2023-07-14 by kbx81

kbx81, 2023-07-14:
> I can enable/disable valve via valve_switch_id state but there aren't any method to test this state.

`state` is not how you would enable/disable...anything. `state` _is_ the current state of the (in this case) `switch`. You'd use the `turn_off()`, `turn_on()`, `toggle()` or `publish_state(new_state)` methods to change the state of the switch.

If you want to check/test the state of a switch, you'd do something like:
```c
if (id(my_switch).state) {
  // stuff to do when switch is 'on'
} else {
  // stuff to do when switch is 'off'
}
```