esphome/feature-requests #2321: DWIN HMI display support?

BrownC0de, 2023-07-22:
Good afternoon, I discovered excellent HMI displays (https://www.dwin-global.com/) that are superior to others (nextion example) both in price and in characteristics (color, resolution, viewing angle, touch sensor) and appearance. But they are quite problematic in setting, from the side of the esphome. Is it possible to finish off their support, and facilitate customization by the user?

Devices with high-quality displays that do not load esp

Unfortunately, these beautiful displays are not currently usable on anything without programming knowledge. (At the same time, setting up the interface itself is not a problem, it has a rather convenient interface), but its communication through the UART with the ESP is not very easy and not pleasant :)

BrownC0de, 2023-07-23:
By the way, the manufacturer DWIN is very willing to provide screens for development and testing.

arnaudcart, 2023-11-15:
Any news on that? I would be interested as well. Thanks!

pdacity, 2023-11-15:
I am also requesting direct support for DWIN displays

O-X-I-G-E-N, 2023-11-15:
Yes, there are really not enough screens!  I'm also looking forward to progress in this direction.

nagyrobi, 2023-11-15:
Which screen models are we talking about here? They offer Android and Linux-based screens also. Which model is ESP8266-based?
Also I see most of the screens require further case development - but are there any screens witch complete housing, which could be interest for ESPHome?

nagyrobi, 2023-11-15:
Wow, the TC040C12U00 is indeed very appealing, and price-wise is also not bad. Seen even on Aliexpress.

BrownC0de, 2023-11-15:
Screen quality, colors, viewing angles, interface speed and touch screen responsiveness.
https://github.com/esphome/feature-requests/assets/34803844/aa2b7968-5650-4aa2-9b1f-4e80eb8c8b43

nagyrobi, 2023-11-15:
> And this is what it looks like from the inside.

There's an unpopulated slot for a WiFi module. Chances are we could even find an ESP-based module which could fit.

randybb, 2023-11-15:
It is called "wifi-10" and it is an ancient esp8266 https://www.aliexpress.com/i/1005001392531088.html would rather use any esp32 (m5stamp, or any other module, and some wires would do it).

BrownC0de, 2023-11-16:
> Can you take a closer shot of that part of the board?

Unfortunately it won’t work out, this is not my photo.
I have screens without housings, there is a different board, with pins simply displayed
![photo_2023-11-15_16-28-07](https://github.com/esphome/feature-requests/assets/34803844/03cd
![photo_2023-11-15_16-17-25](https://github.com/esphome/feature-requests/assets/34803844/7a20250b-91bd-429e-b6
[image]
d7-223966beb8ba)
d4e3-fd05-4e67-8639-aefe092ab5a6)

> It is called "wifi-10" and it is an ancient esp8266
 
Yes, it's very similar.
For esp32 you can use the board HDL662S there is a UART or do without this board and connect the cable via connector FPC\FFC 50Pin.

pkejval, 2023-11-28:
I'm considering switching from Nextion to DWIN because of better price and smaller dimensions of boards. 
Does anyone have it working with ESPHome? Is it difficult to make it work with UART commands only?

hayschan, 2024-01-19:
> I'm considering switching from Nextion to DWIN because of better price and smaller dimensions of boards. Does anyone have it working with ESPHome? Is it difficult to make it work with UART commands only?

They have an [Arduino Library](https://github.com/dwinhmi/DWIN_DGUS_HMI) for DWIN DGUS T5L HMI Displays.

edwardtfn, 2024-01-23:
> By the way, the manufacturer DWIN is very willing to provide screens for development and testing.

I would be happy to work on an ESPHome component for this.
Which model is more popular here?

BrownC0de, 2024-01-23:
> I would be happy to work on an ESPHome component for this.
> Which model is more popular here?
It would be great. СOF products (https://www.dwin-global.com/cof-uart-lcd-module/) It is best suited for use with esphome. I use DMG48320F035_01WTC In my opinion, this is the most successful diagonal, it is also available with a body. I can take part in testing.

edwardtfn, 2024-03-07:
I finally got my displays. I will try to find some time to play with this.

u20p17, 2024-06-12:
any news on this topic? :)

almirus, 2025-10-07:
hi! any news?